vitormhenrique/OctoPrint-Enclosure #165: DHT11 sensor. wont work

Guitarans, 2018-07-07:
Well. ive been running a octopi version that was a bit legacy.. Bit it worked... and then i didnt bother to update it. 
But after some update issues ive decided til bake a fresh install or image on my raspberry pi 2..
But after that i can not get my DHT 11 sensor to work. Ive done the instructions  about the adafruit things. and updated and upgraded the pi.. and even installed the latest PIP...
My GPIO outputs works fine ( ive got some relays hooked up to some pins ) And im only running Enclosure plugin ( i got no other plugin that are using the gpios )

Any good idears ? Thanks

vitormhenrique, 2018-07-07:
Disable sudo on the advanced sections...

Guitarans, 2018-07-07:
My Hero!

Thanks alot... I use your plugin alot. Is there anyhow i can give a bear or two in some kind of donation?.

Best regards from Denmark

Guitarans, 2018-07-08:
> Nice, you will enjoy the new version, there are a lot of new features, A
> LOT! especially using plugin g-codes to control outputs, in conjunction
> with buttons you can do a ton of different stuff, I have a bunch of buttons
> now to automate my printers, change filaments, turn on and off lights etc...
>
> There is a donation button on the main page of the read-me, another way to
> contribute is sending-me different types of sensor that I could implement
> on the plugin to improve it, gift cards for online stores that sell the
> sensors also work (adafruit / sparkfun / amazon)... people have done this
> on the past, when they wanted new sensors implemented that I didn't have in
> hands...
>
> Having new ideas and useful feature requests are always welcome...

Great ill look forward to that.
Well, i dont know if it is a new feature. But i would like to have a
physical button to power off the pi..
Im using a little python script on my other raspberry pi's to power them
off.

vitormhenrique, 2018-07-08:
you can do this already, set one output as "shell script" and put the code as `shutdown –h now` 
You can them add one input as output control link it to this `shell script output`, when you press the button it will turn off the pi.