vitormhenrique/OctoPrint-Enclosure #168: Add pause function to API

jlg89, 2018-07-12:
I can pause a print from a shell script using the OctoPrint API, like so:
echo '{"command": "pause" , "action": "pause"}' | curl -X POST -d @- -H "X-Api-Key: "$apikey"" -H "Content-Type: application/json" http://localhost/api/job

However, that doesn't send me an IFTTT notification! I want to be able to use the Enclosure Plugin API to generate the pause command, so that the IFTTT notification will happen. Can this be added? Or is there already a way to do this, that I'm missing?

vitormhenrique, 2018-07-12:
Notification when paused it's not implemented yet, but my I ask why would it be needed? What I mean is, if you are pausing a print, using a button, or something else, why would you want a notification? you just pressed a button, so you know that you paused the print....

jlg89, 2018-07-12:
It's related to the filament motion sensor. I have a shell script that monitors an input for pulses, with a configurable timeout that will pause the print (via OctoPrint API) if pulses stop for longer than the timeout. I could kludge something together by driving a GPIO output with the script, tying that to an input that's monitored by the plugin, but the API would be cleaner.

vitormhenrique, 2018-07-12:
The filament sensor part of the plugin already has a notification system, instead of building a Frankenstein to make it work it would be much better to implement on the plugin itself....

We have a couple of options:

- You could do whatever implementation in python, for the encoder filament sensor, and send it to me so I can port to the plugin.

- You could wait for me to do all the work and implement the encoder filament sensor part.

- You could study the plugin architecture, implement and make a pull request.

- You can make your implementation with shell scripts call another shell script or even concatenate the shell script to make ifttt to notify you, not that difficult, check this [here](https://github.com/hossman/ifttt-trigger)

Keep in mind that I have a full time job and I work on this plugin as much as I can, but previous releases  took me almost 2 months of weekends and late nights, I completely changed the architecture to make the plugin scalable and implemented a lot of new features. That is the only reason why I haven't invested more time on this new feature. I need to get more projects going, and there is only so much time that I have out of my full time job and other hobbies.

jlg89, 2018-07-12:
The command-line stuff for IFTTT may be the best option...I hadn't seen that info. I can then pause the print with the existing OctoPrint API, and send an IFTTT notification using the same config I'm using for the Enclosure Plugin. That will get me a functional filament motion detector, and maybe one of these days you can work it into the Enclosure Plugin. :)