vitormhenrique/OctoPrint-Enclosure #169: Be RESTUful

bistory, 2018-07-13:
Hi,

First thanks for your work on this plugin, it is awesome and I use it everyday !
I started to look at integrated it to Home-Assistant but it looks like the format of the API is not RESTful at all :(
Here is the format it should adopt to be compliant :

Create a endpoint /plugin/enclosure/outputs to replace /plugin/enclosure/getOutputStatus
Create /plugin/enclosure/outputs/{OUTPUT_ID} to replace /plugin/enclosure/setIO
=> When sending a GET request to this endpoint, it should send the status of the output
=> When sending a POST request to this enpoint, it should set the status out the output

The same applies to /plugin/enclosure/setEnclosureTempHum
Create /plugin/enclosure/temperature/{OUTPUT_ID}
POST will set the value

And so on with other endpoints :)

It would be cleaner and more standard, that will make it work with every RESTful clients natively.
It doesn't seem to be a difficult mod so I could make a pull request but I would like to be sure you agree with these changes :)

Thanks

vitormhenrique, 2018-07-13:
Hello @bistory 
I have no problem with the proposed changes, the current implementation was actually someone's pull requested kind of hacked together, to use as API, the initial implementation was not "thought" to be used as an API.
If you want to work on it go for it! I can help with code review and testing.
Right now I don't have the time to stop and actually implement everything, but I don't have anything against the proposed changes.

Baumi1245, 2018-12-12:
I would also need a REST-Api for the temperature-readings of my DHT22s to impliment them to openHAB.

Thank you....

anthonyst91, 2018-12-12:
The API are not REST-ful but it is not really a problem.
You can get the whole plugin configuration from [http://docs.octoprint.org/en/master/api/settings.html](/api/settings)

and retrieve the updated informations from the websocket (/sockjs/websocket). The different messages look like the following:

{"plugin": {"data": {"rpi_output_neopixel": [{"color": "rgb(0,0,0)", "auto_shutdown": false, "index_id": 15, "auto_startup": false}], "rpi_output_pwm": [], "rpi_output_temp_hum_ctrl": [{"status": true, "auto_shutdown": true, "index_id": 16, "auto_startup": false}, {"status": false, "auto_shutdown": true, "index_id": 17, "auto_startup": false}], "rpi_output_regular": [{"status": 1, "auto_shutdown": true, "index_id": 1, "auto_startup": true}, {"status": false, "auto_shutdown": true, "index_id": 9, "auto_startup": true}, {"status": 1, "auto_shutdown": true, "index_id": 10, "auto_startup": true}]}, "plugin": "enclosure"}}

{"plugin": {"data": {"sensor_data": [{"index_id": 2, "temperature": 0.0, "humidity": 0.0}]}, "plugin": "enclosure"}}

{"plugin": {"data": {"set_temperature": [{"set_temperature": 124.0, "index_id": 16}, 124.0, {"set_temperature": 30.0, "index_id": 17}, 30.0]}, "plugin": "enclosure"}}

{"plugin": {"data": {"filament_sensor_status": []}, "plugin": "enclosure"}}

Baumi1245, 2018-12-12:
Sry... can´t find any data here: (http://>IP</api/settings?apikey=>KEY<)
Temperature at the WEB-UI is showing up correct.

`

2 |  
-- | --
action_type | "output_control"
controlled_io | 1
controlled_io_set_value | "low"
ds18b20_serial | ""
edge | "fall"
filament_sensor_enabled | true
filament_sensor_timeout | 120
gpio_pin | "17"
index_id | 5
input_pull_resistor | "input_pull_up"
input_type | "temperature_sensor"
label | "Input 5"
printer_action | "filament"
temp_sensor_address | ""
temp_sensor_humidity | 0
temp_sensor_navbar | true
temp_sensor_temp | 0
temp_sensor_type | "22"
use_fahrenheit | false
`

Baumi1245, 2018-12-12:
ok... something strange:
the second sensor is showing 24.7 ° at /api/settings

if i refresh the web-ui the data from /api/settings is shown first (0° and 24.7°) after a short time the correct temperature is appearing in the web-ui.

data in /api/settings is not changing.

PR #261 (2019-10-20): Make API RESTful (kind of)
Hi,

I've come up with a first draft on how the API could like so it gets a little more easy to query it. It now also can return temperature and filament inputs. I hope it matches your idea of the API.

I tried sticking to the general API conventions as well closes #169.

@bistory @Baumi1245 could you please check if this matches your requirements.

**NOT TESTED YET**
I haven't tested it yet. I will do that most likely tomorrow if I have access to the OctoPrint System again. Just want to open this here so we can get some feedback.
closed 2019-10-20 by vitormhenrique