vitormhenrique/OctoPrint-Enclosure #170: I2C PCA9685 Output?

Ch3fk0ch88, 2018-07-14:
Hello,
is it possible to control a PCA9685 (PWM FAN CONTROL) as Output?

Thanks

Ch3fk0ch88, 2018-07-16:
Its not connected directly to the Raspberry.
I use 4PIN PWM Fans (PC Case Fan's like Noctuna).
So u connect the 12V and the GND to a DC Power Supply and the PWM and Speed signal to the Raspberry.
But the Raspberry PWM signal has not enough current to run bigger PWM Fans.

vitormhenrique, 2018-07-16:
same thing apply: you can just use the 5v from rpi supply and a transistor to increase the current capabilities of rpi pwm...
Also, those fans like to be controlled by a 25kHz signal, default value on the plugin is set to 50Hz.
Your issue can be easily fixed with hardware. 
I could implement PCA9685 on the future, but it is not going to be a near future, there are other things on my priority list and don't have a PCA9685 on hand to implement and test the feature.

Ch3fk0ch88, 2018-07-16:
Did u maybe auf time to make a little circuit diagram?
This PWM stuff isn't really my area of expertise. 

Otherwise I use as an alternative simply a 2-3Pin fan and control the voltage via a MOSFET which is controlled by the Raspberry PWM^^

Ch3fk0ch88, 2018-07-22:
buh