vitormhenrique/OctoPrint-Enclosure #171: Temp seems to work - but does not show in OctoPrint

Idus, 2018-07-22:
I must start with exposing my throat as a beginner when it comes to RPi - so what might be obvious to you guys, is top science to me. ;)

Here goes...
I am having issues with the **_input a temperature sensor_** - all output works just fine

When I have installed OctoPi - I install/upgrade the Python-script according to you guide and test it out. It tests out fine and give me numbers back - as seen on

[image]

I surf into OctoPrint and set it up - using GPIO pin 4 as default (and for the above test).

[image]

I restart the whole system - but it never shows any feedback in the system if you look at:

[image]

My conclusion is that the sensor is setup correct (since I can get the feedback numbers in the command-prompt) - but that the OctoPrint system interface does not get it.

I´ve had this issue for some time - ever since I changed to OctoPrint 1.3.8 and later versions.

Currently made system from OctoPi 0.15.1 with updated OctoPrint 1.3.9rc4 (need rc-version to even work with my latest Prusa MK3 firmware (v.3.3.1).

And yes, by just installing a clean system (which I have done at least 8-9 since it started to fail) - it consistantly fails with the above settings and feedback.

Cheers and thanx for a great plugin.

Idus, 2018-07-23:
Excellent! That did the trick! Now it shows just fine!

closed 2018-07-23 by Idus

vitormhenrique, 2018-07-23:
all the information that you needed was on the troubleshooting guide....
Glad that it worked out for you...