vitormhenrique/OctoPrint-Enclosure #172: 24 neopixel ring not working

madhunm, 2018-07-26:
I'm having an issue with running a 24 neopixel ring in direct mode.

I'm using a Pi 2

The python examples work fine; do you need access to my pi?

vitormhenrique, 2018-07-26:
Well, I need so many things... but for helping you mostly:
- You octoprint log,
- What exactly have you tried to do to fix the issue,
- What python examples code are you using.

There is a template for creating issues and asking help, there is also an troubleshooting guide, you did not follow any of those.
Also, I made this plugin as a free software, opensource and I support and help users just because I'm nice. But I don't like the assumption that I would stop whatever I'm doing to log on a remote rpi and fix some probably PEBCAK issue. You need to be cordial and follow community guidelines.

Please attach your octoprint log print screens of you configuration, and link to an example code that works on your setup, with those 3 things I'll probably be able to point it out what is going on....

madhunm, 2018-07-26:
@vitormhenrique My sincere apologies if I did not appear to be cordial.

Here are the steps I tried: 

1. Went through the issue list and say this: https://github.com/vitormhenrique/OctoPrint-Enclosure/issues/143#issuecomment-394188491

When I run it with `sudo python test.py 18 60 255 255 255 255 0`, all the lights come on white and seem normal. no flickering, no led dimmer than the other etc.

2. Installed rpi_ws281x and ran the strandtest.py; all the colors are working fine.

Here are the screenshots of my setup:

[image]
[image]
[image]
[image]

About the log; Where is it located?

I once again apologize if I sounded ungrateful;

Thank you.

vitormhenrique, 2018-07-26:
No worries, sometimes I get a little frustrated with lack of information when asking help, you you are all right! 

You can get the octoprint logs following [this](https://discourse.octoprint.org/t/where-can-i-find-octoprints-and-octopis-log-files/299)

But you may try to disable "sudo" under the advanced section of the plugin and try again to use the neopixel.

vitormhenrique, 2018-07-26:
if you have a webcam enabled pointing to your led strip I can remote login on your pi and help troubleshoot, but I would need to see the light turning on / off to test it out...

Shoot an email on the email linked on my github profile with the access info.

madhunm, 2018-07-26:
Here is the Octoprint log:

[octoprint.log](https://github.com/vitormhenrique/OctoPrint-Enclosure/files/2233404/octoprint.log)

I shall revert with the access details in the morning; its 0130 here :)

madhunm, 2018-07-31:
@vitormhenrique I just did. Apologies for the delay.

vitormhenrique, 2018-08-06:
Just to document: there were couple things that I had to do to fix this on @madhunm environment. 
rpi_ws281x really needs sudo, and you need to setup up so your rpi does not ask for a password for sudo, so run:

`sudo visudo`

and add `pi ALL=(ALL) NOPASSWD: ALL` to the end of the file.

I don't know why but this system was refusing to work when passing the DMA instead of hard coding, so I needed to change the default code that handle the neopixel:

`nano /home/pi/oprint/lib/python2.7/site-packages/octoprint_enclosure/neopixel_direct.py`

changed to:

<pre><code>
from neopixel import *
import sys
import time

LED_INVERT = False
LED_DMA = 10
LED_FREQ_HZ = 800000

if len(sys.argv) == 8:
    LED_PIN = int(sys.argv[1])
    LED_COUNT = int(sys.argv[2])
    LED_BRIGHTNESS = int(sys.argv[3])
    red = int(sys.argv[4])
    green = int(sys.argv[5])
    blue = int(sys.argv[6])
    address = int(sys.argv[7], 16)
else:
    print("fail")
    sys.exit(1)

strip = Adafruit_NeoPixel(LED_COUNT, LED_PIN, LED_FREQ_HZ, LED_DMA, LED_INVERT)
strip.begin()

color = Color(green, red, blue)

for i in range(LED_COUNT):
    strip.setPixelColor(i, color)

strip.show()

print("ok")</code></pre>

closed 2018-08-06 by vitormhenrique

madhunm, 2018-08-06:
@vitormhenrique Thanks a million for your help! much appreciated.

sfgabe, 2018-08-26:
For reference, I also had to make the above code changes to get a neopixel 12 ring working on a pi 3b. Works fine now. Thanks @vitormhenrique !

dionmindhive, 2018-10-05:
saved my bacon too. Thanks @vitormhenrique Lé Solved!