vitormhenrique/OctoPrint-Enclosure #173: Running shell script causes web interface to stop responding

jlg89, 2018-07-31:
I created a Shell Script output that runs the filament motion sensor script in feature request. That script generates output to stdout, which is captured in some way by the Enclosure Plugin, and displayed when the script exits. This particular shell script runs for the duration of the print, and it seems that while the Enclosure Plugin is capturing stdout from the script, it causes the OctoPrint web interface to stop accepting connections.

* If I pull up a terminal session and kill the script, the web interface starts responding again.
* If I run the script from the command line instead of using the Enclosure Plugin button, there is no effect on the web interface.
* I have verified that the script does not consume significant CPU or memory resources.

**To Reproduce**
Steps to reproduce the behavior:
1. Create a shell script that doesn't exit, but generates ongoing stdout.
2. Create an Enclosure Plugin "Shell Script" output that runs the script.
3. Start a print, then use the EP button to run the script.
4. Try to reconnect to the OctoPrint web interface.

**Expected behavior**
The web interface should continue accepting connections.

**Additional context**
I didn't have debug enabled, and the printer is in the middle of a large print. Once it's done, I will do further testing and get a log file.

vitormhenrique, 2018-07-31:
That is not an issue, it's by design, the plugin expects the script to run and return a value so it knows that it was executed. If your script blocks or runs for longer you may want to execute it on the background punting an "&" on the end of the command line.

`sudo FilamentMonitor.sh &`

However keep in mind that the enclosure plugin will not be able to communicate with the executing program after you tell it to run. So it will not know when the print finishes.

The correct way to implement what you are trying to do is to create a daemon (service) with your code, and make that daemon accept commands from the bash, so you will always have a "filament monitor" service running on your pi, and you would for example accept commands like `filamentmonitor start`, `filamentmonitor stop` etc....

closed 2018-07-31 by vitormhenrique

jlg89, 2018-07-31:
...or convert it to Python like you've requested, so it would be easier to implement in a plugin. :/

vitormhenrique, 2018-07-31:
yes, or that! =P