vitormhenrique/OctoPrint-Enclosure #175: HTU21D

Shalvan, 2018-08-02:
Hi, 
I'm wondering what do You think of adding HTU21D sensor to Your plugin? 
I know it was mentioned earlier issues... And You sad that You dont have this sensor... 
Ok the point... Looking at the datasheet of HTU21D and Si7021 they at the same, same commands, same bits etc. 
But when i connected it to RasPi and run Si7021 script directly in the cmd Line i am getting - 1 | - 1
I went throu your Code and seen that this value represents bad address... Maybe i am running the Code wrong? Should i Add address at the end of execution command? Like ./Si7021.py 40?
Tje address of both HTU21D and Si7021 is the same 0x40
I've also tried different Code witch was working in cmd Line but not in browser :/ 
(before trying browser i recompiled Python program). 
What do You think? Maybe together we can solve the problem and it will be easier to add that sensor :)

vitormhenrique, 2018-08-02:
Yes, you need to pass the address to the script.
`python Si7021.py 40`

If that works, you don't need to change anything anywhere, just choose Si7021 as the sensor, and put address 40.

Shalvan, 2018-08-03:
I've added the address at the end of the line but that gave me an error:
IOError: [Errno 121] Remote I/O error
So Your code only works with SI7021 ( as You mentioned ).
I have working code for HTU21D but i dont know how to send the vaules to Your pugin

vitormhenrique, 2018-08-06:
There is no way to use an external script today, I can add this feature on the next release, now, you could for example override the existing code on the SI7021 with your code and it would probably work.....

Shalvan, 2018-08-10:
Here i have python file that is working form console and should output the same data format as Your script 
`#!/usr/bin/python
import struct, array, time, io, fcntl

I2C_SLAVE=0x0703
HTU21D_ADDR = 0x40
CMD_READ_TEMP_HOLD = "\xE3"
CMD_READ_HUM_HOLD = "\xE5"
CMD_READ_TEMP_NOHOLD = "\xF3"
CMD_READ_HUM_NOHOLD = "\xF5"
CMD_WRITE_USER_REG = "\xE6"
CMD_READ_USER_REG = "\xE7"
CMD_SOFT_RESET= "\xFE"

class i2c(object):
   def __init__(self, device, bus):

          self.fr = io.open("/dev/i2c-"+str(bus), "rb", buffering=0)
          self.fw = io.open("/dev/i2c-"+str(bus), "wb", buffering=0)

          # set device address

          fcntl.ioctl(self.fr, I2C_SLAVE, device)
          fcntl.ioctl(self.fw, I2C_SLAVE, device)

   def write(self, bytes):
          self.fw.write(bytes)

   def read(self, bytes):
          return self.fr.read(bytes)

   def close(self):
          self.fw.close()
          self.fr.close()

class HTU21D(object):
   def __init__(self):
          self.dev = i2c(HTU21D_ADDR, 1) #HTU21D 0x40, bus 1
          self.dev.write(CMD_SOFT_RESET) #soft reset
          time.sleep(.1)

   def ctemp(self, sensorTemp):
          tSensorTemp = sensorTemp / 65536.0
          return -46.85 + (175.72 * tSensorTemp)

   def chumid(self, sensorHumid):
          tSensorHumid = sensorHumid / 65536.0
          return -6.0 + (125.0 * tSensorHumid)

   def crc8check(self, value):
          # Ported from Sparkfun Arduino HTU21D Library: https://github.com/sparkfun/HTU21D_Breakout
          remainder = ( ( value[0] << 8 ) + value[1] ) << 8
          remainder |= value[2]

          # POLYNOMIAL = 0x0131 = x^8 + x^5 + x^4 + 1
          # divsor = 0x988000 is the 0x0131 polynomial shifted to farthest left of three bytes
          divsor = 0x988000

          for i in range(0, 16):
                 if( remainder & 1 << (23 - i) ):
                        remainder ^= divsor
                 divsor = divsor >> 1

          if remainder == 0:
                 return True
          else:
                 return False

   def read_tmperature(self):
          self.dev.write(CMD_READ_TEMP_NOHOLD) #measure temp
          time.sleep(.1)

          data = self.dev.read(3)
          buf = array.array('B', data)

          if self.crc8check(buf):
                 temp = (buf[0] << 8 | buf [1]) & 0xFFFC
                 return self.ctemp(temp)
          else:
                 return -255

   def read_humidity(self):
          self.dev.write(CMD_READ_HUM_NOHOLD) #measure humidity
          time.sleep(.1)

          data = self.dev.read(3)
          buf = array.array('B', data)

          if self.crc8check(buf):
                 humid = (buf[0] << 8 | buf [1]) & 0xFFFC
                 return self.chumid(humid)
          else:
                 return -255

if __name__ == "__main__":
        # Create humdity sensor object
        obj = HTU21D()
		# Read temp and humidity and log to file
        temp = obj.read_tmperature()
        humid = obj.read_humidity()
        # out_string = "Temp=%.1f Humi=%.1f" % (temp, humid)
        #out_string = ('{0:0.1f} | {1:0.1f}'.format(temp, humid))
		# out_string = "%.1f | %.1f" % (temp, humid)
		# out_string = "Temp=%.1f Humi=%.1f" % (temp, humid)
        # print out_string
        print('{0:0.1f} | {1:0.1f}'.format(temp, humid))
        time.sleep(1)
		
		`

I've replaced Your file with this and compiled all python files again 
but still no response
Maybe the reason is that plugin is adding address at the end of execution command?
I don't know how to get it working inside plugin :(

vitormhenrique, 2018-08-10:
if you are hacking the file you need to follow the same format as mine, and accept the parameters that I pass (I'm passing the address to the command).
I'll create a feature enhancement on the future to accept external scripts to read the temperature, for now, I suggest you just using a supported sensor, there are so many already, and most are very very cheap.

reshuler, 2018-09-06:
Hi, 

I followed your file format and made the changes below, which works for HUT21D, SI7021, and SHT21.  I expect other similar parts to these will work also.  I used information from "https://github.com/jaques/sht21_python.git" to create this mod.  This code doesn't use smbus.  I found an issue when I trying to read bytes back to back from smbus.  This why I created this version.

Regards,
Robert

import fcntl
import time
import sys

SOFT_RESET = 0xFE
TEMP_NO_HOLD = 0xF3
HUM_NO_HOLD = 0xF5

I2C_SLAVE = 0x703
I2C_SLAVE_FORCE = 0x0706
I2C_DEVICE_NUM = 1

STATUS_BITS_MASK = 0xFFFC

TEMP_WAIT_TIME = 0.086
HUM_WAIT_TIME = 0.030

if len(sys.argv) == 2:
    address = int(sys.argv[1],16)
else:
    print('-1 | -1')
    sys.exit(1)

i2c = open('/dev/i2c-%s' % I2C_DEVICE_NUM, 'r+', 0)
fcntl.ioctl(i2c, I2C_SLAVE, address)
i2c.write(chr(SOFT_RESET))
time.sleep(0.050)

i2c.write(chr(TEMP_NO_HOLD))
time.sleep(TEMP_WAIT_TIME)
data = i2c.read(3)

unadjusted = (ord(data[0]) << 8) + ord(data[1])
unadjusted &= STATUS_BITS_MASK  # zero the status bits
unadjusted *= 175.72
unadjusted /= 1 << 16  # divide by 2^16
unadjusted -= 46.85

cTemp = unadjusted

i2c.write(chr(HUM_NO_HOLD))
time.sleep(HUM_WAIT_TIME)
data = i2c.read(3)

unadjusted = (ord(data[0]) << 8) + ord(data[1])
unadjusted &= STATUS_BITS_MASK  # zero the status bits
unadjusted *= 125.0
unadjusted /= 1 << 16  # divide by 2^16
unadjusted -= 6
humidity = unadjusted

if humidity > 100.0:
   humidity = 100

print('{0:0.1f} | {1:0.1f}'.format(cTemp, humidity))

i2c.close()

Shalvan, 2018-09-20:
Hi there Reshuler,

I try'd Your code, and after running sudo python filename.py 40 i get nothing . If i dont write address at the end of the command i'm getting -1 | -1 value
Do You know what can be a problem ?

reshuler, 2018-09-20:
Hi Shalvan,

Please send me the code  you are using, information the temperature module you are using, and the I2C address of the temperature module you are using.

The software mods are working well on my side.

Are you sure you are able to talk to the Temperature module on the I2C bus?  I used a logic analyzer to verify I had the correct I2C address before I modified the software. 

Have you run sudo i2cdetect -y 1 to verify you temperature module is being seen on the I2c bus?

Thanks,
Robert

Shalvan, 2018-09-20:
I ran i2cdetect -y 1 and i see that my device is at address 40
for testing i used code i posted one post above Yours, and that piece of code is working
The module i am using is HTU21D

reshuler, 2018-09-20:
Hi Shalvan,

It is good that you see the module at address 0x40.  I’m also using a similar module.

Can you run the python code outside of the enclosure environment and see what happens?

Thanks,
Robert

Shalvan, 2018-09-20:
This is the exact output for 0x40

pi@octopi:~/oprint/lib/python2.7/site-packages/octoprint_enclosure $ sudo python SI7021.py 0x40                          pi@octopi:~/oprint/lib/python2.7/site-packages/octoprint_enclosure $

for only 40
pi@octopi:~/oprint/lib/python2.7/site-packages/octoprint_enclosure $ sudo python SI7021.py 40                          pi@octopi:~/oprint/lib/python2.7/site-packages/octoprint_enclosure $

and for nothing in arguments
pi@octopi:~/oprint/lib/python2.7/site-packages/octoprint_enclosure $ sudo python SI7021.py
-1 | -1
pi@octopi:~/oprint/lib/python2.7/site-packages/octoprint_enclosure $

reshuler, 2018-09-20:
Hi Shalvan,

Pass it a 64.  It converts the second arguments to hex.

Regards,
Robert

reshuler, 2018-09-20:
First argument the program name, the second argument is the value.

I would recommend that you add print statements in the code and see where the failure is.   -1 | -1 is the first check/failure for passed arguments.  If you are getting pass that, then you should be able to add prints in your code at various points to check the values at various points.

I’m using a HUT21 module I got off amazon, and I have it working well. 

Regards,
Robert

Shalvan, 2018-09-20:
so it seams that after 

sys.exit(1)

program stops... Just before i've made a print statment and it was priting ok... i've made another print after sys.exit(1) and ... bam .. nothing

reshuler, 2018-09-20:
When i pasted the code into github, the spacing format wasn’t kept.  The sys.exit(1) should be on the same space level the print in the line above.

The code should look like:
if len(sys.argv) == 2:
   address = int(sys.argv[1], 16)
else:
   print(’-1 | -1’)
   sys.exit(1)

if sys.exit(1) is not at the same space (indention) as the print above, the code will always quit.

Regards,
Robert

reshuler, 2018-09-20:
The same thing for the humidity if.  Needs spaces or humidity will always be 100.

Regards,
Robert

Shalvan, 2018-09-20:
now its getting better :)

now i have 
pi@octopi:~/oprint/lib/python2.7/site-packages/octoprint_enclosure $ sudo python SI7021.py 64
Traceback (most recent call last):
  File "SI7021.py", line 26, in <module>
    i2c.write(chr(SOFT_RESET))
IOError: [Errno 121] Remote I/O error

reshuler, 2018-09-20:
Much better.  Can you add code to print out the address after it is converted to hex, so we can verify the address.

I’ve had the Error 121 before.  It is a basic communication error message.  How long is your cable to the module?  I2C see is sensitive to length.  You may want to try the module on a short cable connected up the Raspberry Pi.

Regards,
Robert

Shalvan, 2018-09-20:
I've edited my post ... now it seams to work in console 

pi@octopi:~/oprint/lib/python2.7/site-packages/octoprint_enclosure $ sudo python SI7021.py 0x40
25.3 | 56.0

but in web interface its not working maybe i need to reset but i have ongoing print atm, 
ive did python -m compileall .       after all changes in the file but i think i still need to reboot the server

reshuler, 2018-09-20:
Good to see that you got it working.  Sorry for the bad format of the source file.

You can compile the code, but Octopi may do it for you on start up.

You need to make sure you pick the SI7021, with an address of 0x40, and the correct I2C interface in the plug in.

How did you get pass the Error 121?

Regards,
Robert

Shalvan, 2018-09-20:
I've made code formatting ( all the spaces ) then  i've tried different addresses in console and found that 0x40 is working 
now i need to test the web interface, at the moment it isn't working. mayne restart will help, but i will be able to do so after the print. so tomorrow i will send resoults here :)

Much apriciate the help ;)

reshuler, 2018-09-20:
Good Luck.

Thanks,
Robert

Shalvan, 2018-09-21:
Ok,
So after formatting the Code aka adding few spaces, finding proper device address witch was 0x40 and disabling sudo check box in the plugin everything is working perfect. 
BIG THANKS for the perfect Code ;) Reshuler 
And also BIG to Vitormhenrique for creating this awesome plugin.

reshuler, 2018-09-21:
Great.

Thanks,
Robert

januar3829, 2019-02-13:
> box

How did you get it to work in the enclosure Plugin? 
After copy&past the code above the script runs perfekt and returns correct data, but at the octopi website I still see 0°C and 0%

vitormhenrique, 2019-02-13:
Uncheck use sudo on advanced section.

januar3829, 2019-02-14:
> Uncheck use sudo on advanced section.
I did uncheck that box with no effect. Several times to be exact, but this time it worked! Thank you! 
So next Question is, why can't I change the Adress of that Sensor in order to use multiple ones :(

QuestionDevelopment, 2021-05-19:
Just a note to everyone...I couldn't get this to work in the plugin (no problem via cli) until I hardcoded the address.  Not sure why but I tried all different values in the plugin settings and none would work.