vitormhenrique/OctoPrint-Enclosure #176: Filament Splicer/Fuser/Joiner

DaveDischord, 2018-08-06:
With all the features/choices available, I'm assuming its already possible, but to clarify, would it be possible to control what amounts to a stand-alone hotend, heating element and thermistor only, using the Enclosure plugin?  I've found controllers, such as this [Filament Fuser Controller
](https://www.tindie.com/products/JasonKits/3d-printer-filament-fuser-filament-joiner-kit/), but figured it'd be nice to have my OctoPi do that have it close to my printer.

vitormhenrique, 2018-08-06:
no, you can not directly control a hot end, this plugin only controls the gpio from the raspberry pi. 
but when I mean you can't you could, with some effort, you would need to build a drive circuit to the hot end, and a temperature sensor that could accept the temperatures from the hotend, this plugin does not directly read thermistors.  The best way to accomplish what you want is to buy a cheap arduino [ramps shield](https://www.amazon.com/OSOYOO-Printer-Controller-Stepper-Heatsink/dp/B0111ZSS2O/ref=sr_1_3?ie=UTF8&qid=1533569445&sr=8-3&keywords=RepRap+RAMPs&dpID=61tKkvS%252BVvL&preST=_SX342_QL70_&dpSrc=srch), and plug your hotend to it.
Then you could use this plugin on a ri and the arduino to add buttons to automate your work.

closed 2018-08-06 by vitormhenrique

DaveDischord, 2018-08-06:
Awesome!  I've read how that would be an easy solution, using a ramps kit, so I'll work them in together.  Thank you.