vitormhenrique/OctoPrint-Enclosure #177: SI7021

jessefannon, 2018-08-10:
I am attempting to connect the SI7021 to octoprint.

I have enabled the I2C in the settings.  I have tested the sensor with the python file and it appears to be working correctly:
pi@octopi:~/oprint/lib/python2.7/site-packages/octoprint_enclosure $ python SI7021.py 40
25.5 | 56.3

My problem is that the Enclosure Controls page on OctoPrint is showing 0 for both the temp and humidity.  

Do you have any suggestions on what I may need to change to get them to display correctly?

[image]

vitormhenrique, 2018-08-10:
1. enable / disable sudo under the advanced settings. 
2. follow the guideline and attach octoprint log when opening an issue.

jessefannon, 2018-08-13:
My apologies for not following the guideline and attaching the log.  Disabling SUDO has fixed my problem.

Thank you.

closed 2018-08-13 by jessefannon