vitormhenrique/OctoPrint-Enclosure #178: Allow trigger other output with temperature alarm

danijoo, 2018-08-11:
I have a temperature sensor connected that reads room temperature. Additionally, I added a G-code output that sends "M112" to the printer to shut it down. I want to link them together to shut the printer down if a certain temperature is reached

There are two ways this could be archived: 
1) Extend Temperature alarm to send GCODE commands
2) A way to chain outputs together

vitormhenrique, 2018-08-14:
Question, do you have an external relay to control the printer power? or is it implemented on the printer on the pritner firmware ?

danijoo, 2018-08-14:
I want to have it implemented in the printer firmware so it should be triggered by gcode explicitly. 

---- On Tue, 14 Aug 2018 16:51:08 +0200 notifications@github.com wrote ----

Question, do you have an external relay to control the printer power? or is it implemented on the printer on the pritner firmware ?

vitormhenrique, 2018-08-14:
I will put as a FR, but with very low priority, as most users use external relays and the plugin can already shutdown under an alarm based on an external GPIO, and you are probably the only user that will have a custom firmware and board built to accept this current scenario.