vitormhenrique/OctoPrint-Enclosure #179: Neopixel want work

Jacks3DPrinting, 2018-08-19:
Hi,

tryed to controll a neopixel ring via the plugin. But had no success.
If i use the examples from the jgarff´s library it works. 

Set the GPIO pin to 18 like in this tutorial.
https://learn.adafruit.com/neopixels-on-raspberry-pi/software

Iam using a 3B+

OctoPrint: 1.3.9
Enclosure Plugin: 4.12

(https://github.com/vitormhenrique/OctoPrint-Enclosure/files/2300561/octoprint.1.log)

[image]

vitormhenrique, 2018-08-20:
follow this:

https://github.com/vitormhenrique/OctoPrint-Enclosure/issues/172#issuecomment-410864599

Jacks3DPrinting, 2018-08-20:
after reinstalling the python library and sudo visudo it works.

Big thanks awsome plugin :)

closed 2018-08-20 by Jacks3DPrinting