vitormhenrique/OctoPrint-Enclosure #180: Correct state at the status bar when refreshing Octoprint page.

GeorgeVasil, 2018-09-03:
Hello,

First of all congrats for this useful plugin!

I am a newbie user of Octoprint trying to control some LED strips via a relay (simple on/off).

I have enabled "Set button to navbar" and its working great.....BUT when I refresh the octoprint page it reports a wrong state of the relay. To be more specific, assuming that I click ON and then refreshing the page, the navbar button reports OFF. In order to correct the state, I press OFF once (nothing happens), then it goes to on and everything works fine. 

Is there a way to correct that issue?

Thanks for your time,

GV

vitormhenrique, 2018-09-04:
Hello,

The plugin should update the status every 10 or 15 seconds, and self correct it.

Are you sure that even after you wait a bit the status does not change? How about the status on the main enclosure tab?

GeorgeVasil, 2018-09-04:
Hello again and thanks for the fast reply!

The status does not change after 15 sec, nor the status on the main enclosure tab.

I found that if I change the auto-start setting it refreshes to the correct state.

vitormhenrique, 2018-09-04:
That does not make any sense, the update for the ui has nothing to do with the settings, I would need access to your installation over the internet to see what is going on, I can not reproduce the issue on any of my installations.

GeorgeVasil, 2018-09-06:
Could you please send me your mail? I would rather send you access credentials there. 

You need http and ssh to my RPi correct?

vitormhenrique, 2018-09-06:
yeap.... email is on my github page.

vitormhenrique, 2018-11-08:
closing because OP did not provide more info.

closed 2018-11-08 by vitormhenrique