vitormhenrique/OctoPrint-Enclosure #181: Temperature sensor fine tuning

StudioKamar, 2018-09-08:
Hello.
I have two dht11 sensors one on bed level to read temperature acuratelly and second on the top of enclosure to turn on air mixing fan if on top air gets too hot. Problem is my two tht sensors show complettely different temperatures is there a possibility to enter +/- temp to even them out ??

vitormhenrique, 2018-09-09:
probably the sensors will not only differ a fixed amount, but a range on different temperatures, there is no way for do this right now, also, multiple dht sensors are not suggested, they use a lot of cpu resources and fail a lot.
Using i2c sensors or the DS18B20 (even better because you put multiple sensors on the same pin).
You can get DS18B20 sensors for less than a dollar on ebay.

StudioKamar, 2018-09-09:
ok i try that.

closed 2018-09-09 by vitormhenrique