vitormhenrique/OctoPrint-Enclosure #182: auto_shutdown

xXKarneXx, 2018-09-09:
After a few hours of printing, the plugin turns off the printer. But auto shut down is disabled.

2018-09-08 19:00:54,578 - octoprint.plugins.enclosure - INFO - Scheduling regular output id 2_auto_shutdown on 18845.423422 delay_seconds
2018-09-08 19:00:54,584 - octoprint.plugins.enclosure - INFO - Events scheduled to run [{'thread': <_Timer(Thread-646256, initial daemon)>, 'queue_id': '2_auto_shutdown'}]

[octoprint (8).log](https://github.com/vitormhenrique/OctoPrint-Enclosure/files/2364042/octoprint.8.log)

vitormhenrique, 2018-09-09:
picture of the settings please....
I would say remove the output that controls the printer, save the settings and create it again...
Something got messed up on your settings.

vitormhenrique, 2018-09-09:
enable the debug under the advanced sections, delete the current output, save the settings, open the settings again, re-create the outputs and save it again, then send me the print-screens and new log file.

PPPBerlin, 2018-10-31:
In the UI, the setting is disabled, but it is still in the config.yaml. I spend lots of time to solve the problem, that my printer is shutting down every night at 00:20 and start to blaim my power supplier ;) So, please fix this setting and remove the shutdown_time correctly!

vitormhenrique, 2018-11-08:
I can't reproduce your issue @PPPBerlin, if you don't add attach your configuration files, print screen of your configuration there is no way that I can help you.

closed 2018-11-08 by vitormhenrique