vitormhenrique/OctoPrint-Enclosure #183: Temperature and humidity keep reading "0"

bmangodt, 2018-09-09:
I have wired up an AM2302 temperature and humidity sensor to my RPi and installed the correct libraries. When I run the example code, like the readme mentions, the values are correct. However, when I configure the sensor as an input in the plugin, it keeps saying zero... Am I missing something or is this a bug. (See attached images for clarification)

[image]
[image]
[image]

vitormhenrique, 2018-09-09:
if you had done a quick search before opening an issue or read the troubleshoot guide you would see that you probably need to disable sudo on the advanced sections, but i guess is always easier to ask for help than reading the documentation.
It's frustrating spending hours and hours of my free time building something for free, documenting everything and people keep asking the same questions over and over again just for pure laziness...

bmangodt, 2018-09-09:
Look, I've must have just read over that sentence... I really didn't mean to waste your time and it wasn't for pure laziness. The only free time I get for tinkering with this are the weekends and other people like my family also want my attention because they don't see me that much during the week because of work load. So I'm sorry that you feel that way, sorry for my "noobiness" but still thank you for answering my question

vitormhenrique, 2018-09-09:
sorry for the rash words, but I try to document everything really well, wrote troubleshooting guide and spent months and months perfecting the plugin... 
And I get the same question every week or so... =/
I really don't mind helping people out, I even remote login on peoples computer trouble shoot weird issues, fix their setup, there is really not a problem at all being a beginner.... we all were at one point...
But I wish people read the troubleshooting guide first....

bmangodt, 2018-09-09:
I understand it gets frustrating... I should've read the documentation more thoroughly before commenting. I get upset if people keep asking me the same questions at work too, and now I'm guilty of that same offence... Again my apologies!
The issue is resolved now so thank you. I, and the community, appreciate all you do! And it's by far the best plugin for OctoPrint out there!

vitormhenrique, 2018-09-09:
good that it fixed your issue.
have a nice day!

closed 2018-09-09 by vitormhenrique