vitormhenrique/OctoPrint-Enclosure #185: Enclosure plugin tab contains nothing

TTN-, 2018-09-12:
1. I installed the Enclosure plugin.

2. Restarted octoprint.

3. Went to octoprint tab and noted there was nothing there, see screenshot:

[image]

4. I checked the install log. Looks fine.

Attached: 
[plugin_pluginmanager_console.log.txt](https://github.com/vitormhenrique/OctoPrint-Enclosure/files/2374497/plugin_pluginmanager_console.log.txt)

This is octoprint version 1.3.9 this is a rather old install. Probably around 2 years old that has just been updated when updates are available.

If any other info is required please let me know.

vitormhenrique, 2018-09-12:
I don’t understand, have you actually went to settings,  enclosure plugin
and set it up first? It is going to be blank by default.

You have to set it up first.

TTN-, 2018-09-12:
OHHH.

What a face palm moment. It'd be helpful if its empty, it could contain a message saying go to settings, setup the plugin.

sajovicd, 2018-09-12:
Or..... You could read some documentation...

TTN-, 2018-09-12:
I did. I read the readme. Where does it have this information? I read the troubleshooting guide too and searched in closed issues and looked in open issues to find a solution. The answer was not easy to find. In fact I didn't find one at all and that's why I'm posting here. .. just tradin saying.. I tried..

vitormhenrique, 2018-09-20:
@TTN- you clearly missed the Configuration part of the read-me. But that is ok...

closed 2018-09-20 by vitormhenrique