vitormhenrique/OctoPrint-Enclosure #186: Support for smoke detector?

kaju666, 2018-09-18:
Hello

Has anyone connect a smoke sensor to this plugin? for example  MQ-2 Sensor.
It will be a very good feature to cut of the power when smoke is detected.

shbatm, 2018-09-20:
I connected an actual cheap smoke detector as an input using this guide here: https://www.instructables.com/id/Modify-a-6-EUR-smoke-detector-for-use-with-microco/.  Make sure to get a photoelectric detector instead of an ionization one.  I bought my detector for $5 on eBay: https://www.ebay.com/itm/113062422041, 20 minutes of soldering to connect some leads and a resistor and it works to trigger an RPi GPIO input, then I just set it to Cancel Print with the enclosure--tested by pushing the "Test" button on the detector and it cancels the print, (which in my setup then also triggers the printer to kill the power after 60 sec).

vitormhenrique, 2018-09-20:
thanks @shbatm for the answer.

yes @kaju666  you can configure smoke detectors, just read the documentation...

closed 2018-09-20 by vitormhenrique