vitormhenrique/OctoPrint-Enclosure #188: IFTTT event receiving

WarrenSchultz, 2018-09-23:
**Is your feature request related to a problem? Please describe.**
Temperature / humidity triggers from IFTTT for additional data outside of RPi sensors.

**Describe the solution you'd like**
I've got temp/humidity sensors which work with IFTTT, I'd like to be able to poll that data, i.e. receive events from IFTTT triggers to initiate behavior on OctoPrint-Enclosure

WarrenSchultz, 2018-09-23:
See here for the specific model I'm using:
https://www.ambientweather.com/amws0265.html

vitormhenrique, 2018-11-08:
I don't plan to incorporate IFTTT, but I it is on the pipeline to implement MQTT, so it would kind of be possible to link the two together. You would need something to bridge the two, like home assistant, but it would be possible
Check this issue to get status when implemented: https://github.com/vitormhenrique/OctoPrint-Enclosure/issues/40

closed 2018-11-08 by vitormhenrique