vitormhenrique/OctoPrint-Enclosure #189: Incorporate core temperatue (vcgencmd measure_temp) as an input

marshaul, 2018-09-24:
It would be nice if the core temperature could be used as an input, or really for that matter the output of any shell script/command. At the very least it would be convenient to have this information present with the other temps, and it might even be nice to have an option to latch enclosure heaters off in the event core temp exceeds some threshold.

vitormhenrique, 2018-09-25:
it's already on the pipe line using shell scripts as temperature inputs.

vitormhenrique, 2018-09-25:
and you can already set maximum temperature limits where temperature control will be disabled if reached.

[image]

elvinhaak, 2019-10-20:
Hi, I'm looking for the same thing. Has this been added already (and I'm not handy enough to find it) ? I want to control a cooler-fan with the core-temp in mind (and use some other controls like the enclosure-heater at the same time).

Thanks.