vitormhenrique/OctoPrint-Enclosure #190: GPIO Output always HIGH when adding an output on any pin

bzeeman, 2018-09-25:
**Describe the bug**
When I set an output to any BCM pin (or if I change to board pins the board pin) and save it, the GPIO immediately goes HIGH. If I select "Active Low" that is what I would expect to happen, and is what happens. However, If I unselect "Active Low" the GPIO still goes HIGH. 
In the debug log I see `Setting GPIO pin 21 as OUTPUT with initial value: 0` but the PIN goes HIGH anyway.
In fact, no matter what I do, as soon as I add an output for any pin (I have tried 21, 20, and 14) the pin goes HIGH. As soon as I delete the output, the pin goes LOW.

When I toggle the output on or off, I see the following: 
`2018-09-25 03:06:00,824 - octoprint.plugins.enclosure - INFO - Writing on gpio: 21 value True`
`2018-09-25 03:06:04,345 - octoprint.plugins.enclosure - INFO - Writing on gpio: 21 value False`

but the pin stays HIGH the entire time.

I can manually run a python script that turns it on and off without issue.
**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Settings > Enclosure Plugin' 
2. Click on 'Add Outputs...'
3. Add a GPIO output on any available pin (21 is what I use here)
4. Toggle off 'Active Low'
5. Click 'Save'
6. Observe pin 21 goes high immediately after save. (Relay switches on and my lights turn on)
7. octoprint.log output:

`2018-09-25 03:13:25,173 - octoprint.plugins.enclosure - INFO - rpi_outputs: [{u'linked_temp_sensor': u'', u'startup_time': 0, u'temp_ctr_deadband': 0, u'neopixel_brightness': 255, u'new_duty_cycle': u'', u'gpio_pin': u'21', u'default_duty_cycle': 0, u'neopixel_color': u'rgb(0,0,0)', u'hide_btn_ui': False, u'temp_ctr_set_value': 0, u'temp_ctr_default_value': 0, u'default_neopixel_color': u'', u'controlled_io_set_value': u'Low', u'auto_shutdown': False, u'shell_script': u'', u'label': u'Ouput 1', u'toggle_timer': False, u'toggle_timer_off': 0, u'alarm_set_temp': 0, u'auto_startup': False, u'controlled_io': 0, u'shutdown_time': 0, u'temp_ctr_type': u'heater', u'gcode': u'', u'shutdown_on_failed': False, u'temperature_b': 0, u'temperature_a': 0, u'neopixel_count': 0, u'duty_cycle': 0, u'toggle_timer_on': 0, u'show_on_navbar': False, u'duty_b': 0, u'duty_a': 0, u'pwm_status': 50, u'gpio_status': False, u'pwm_frequency': 50, u'startup_with_server': False, u'active_low': False, u'temp_ctr_max_temp': 0, u'pwm_temperature_linked': False, u'temp_ctr_new_set_value': u'', u'output_type': u'regular', u'microcontroller_address': 0, u'index_id': 1, u'new_neopixel_color': u''}]`

`2018-09-25 03:13:25,176 - octoprint.plugins.enclosure - INFO - rpi_inputs: []`

`2018-09-25 03:13:25,179 - octoprint.plugins.enclosure - INFO - Setting GPIO pin 21 as OUTPUT with initial value: 0`
8. Click on "Enclosure Plugin" Tab in Octoprint interface
9. Toggle Output 1  "on"
10. Observe octoprint.log output:
`2018-09-25 03:18:34,808 - octoprint.plugins.enclosure - INFO - Writing on gpio: 21 value True`
11. Toggle Output 1 "off"
12. Observe octoprint.log output:
`2018-09-25 03:18:57,775 - octoprint.plugins.enclosure - INFO - Writing on gpio: 21 value False`

**Expected behavior**
The state of the GPIO should reflect the state of the Enclosure Plugin Output value.

**Additional context**
Raspberry Pi 3 Model B+
`pi@octopi:~/.octoprint/logs $ uname -a`
`Linux octopi 4.14.52-v7+ #1123 SMP Wed Jun 27 17:35:49 BST 2018 armv7l GNU/Linux`

[image]
[image]

vitormhenrique, 2018-09-25:
can you attach the log?

vitormhenrique, 2018-09-25:
the code clearly check the settings and should handle that...

https://github.com/vitormhenrique/OctoPrint-Enclosure/blob/master/octoprint_enclosure/__init__.py#L904

Send me the entire log and I'll figure it out...

vitormhenrique, 2018-09-25:
another thing, the log that you have here says everything is correct: active_low = false, so it would start with 0 and when turning the gpio on it would turn it to true, and when turning it off it would turn it to false.

I have tested this over and over again, and plenty of other users...

bzeeman, 2018-09-25:
> another thing, the log that you have here says everything is correct: active_low = false, so it would start with 0 and when turning the gpio on it would turn it to true, and when turning it off it would turn it to false.
> 
> I have tested this over and over again, and plenty of other users...

I will prepare a log of this occurring. The problem is not that the log is wrong in what it says it's state is, it's that as soon as I add an output on any pin, that pin goes HIGH and stays HIGH no matter what the log output says it is setting things to.
[octoprint.log](https://github.com/vitormhenrique/OctoPrint-Enclosure/files/2416449/octoprint.log)
 In this log you will see the adding of the output, what the software says it is setting the pin out to, me toggling the output on and off 2 times, then deleting the output from settings. I've also attached a video so you can see the behavior of the enclosure lights that I have attached to the relay controlled by GPIO 21

[EnclosurePluginIssue.mp4.zip](https://github.com/vitormhenrique/OctoPrint-Enclosure/files/2416501/EnclosurePluginIssue.mp4.zip)

bzeeman, 2018-09-26:
Because I often make mistakes I decided that I ought to investigate further to see if there was anything on my end causing the issues. I got a new Raspberry Pi 3 B (not plus this time) and downloaded the latest octopi image, flashed that to an SD Card, started up the new RPi with Octopi and set it up as normal. I installed no other libraries, plug-ins, or anything else in either the octopi instance or the general OS. I am getting the same results as before. As soon as I add an output, the pin goes HIGH and does not go low unless I delete the output in the plugin settings.

I also created two python scripts to turn on and off GPIO pins and If I use the Shell Script option on Outputs and run the script it functions properly.

General-IoN, 2018-11-04:
I have the same issue, but with gpio 12
I have tried several things:
Other Raspberry Pi 3B --> no issue on every output
Other Browser --> issue on every output, some freeze high, some low
After adding some debug information to __init__.py i found out that the problem comes from here:
[https://github.com/vitormhenrique/OctoPrint-Enclosure/blob/master/octoprint_enclosure/__init__.py#L188](url)

the value always has the same state
hope this helps

bzeeman, 2019-01-26:
Still having the same issue. It looks like you might have found the place in the code, but I am not sure the developer is going to fix it.

bzeeman, 2019-01-26:
I ended up just writing python scripts to handle what I need and adding buttons to toggle on the light. It would be great if the heater would work.

vitormhenrique, 2019-01-27:
I just got a new raspberry pi, installed everything from scratch, and tested. I have no issue with the GPIOS. There are a LOT of users using this plugin and you guys seam to be the only ones having this issue.
I get really upset with comments like "the developer is not going to fix it". I have a full time job and many other projects, and developed this plugin for months, spent more than 1000 hours coding including trouble shooting and helping people out....
Anyone can troubleshoot, find issues, fix them and send pull requests, I haven't seen any pull requests from you.
I get probably 5 dollars "donations" a year form the button on the main page, that don't even cover the sensors that I buy when people request me to add them.
All that begin said, if you want me to help you, go on my github page, send me an email with access to your octoprint information, you will need to open it on the internet, give me the ip address and password, I'll ssh on your raspberry pi and try to understand what is the issue, I'm glad to help.

closed 2019-01-27 by vitormhenrique

bzeeman, 2019-01-31:
My apologies vitormhenrique. This seems to be an issue that’s system wide. The reason my scripts were working was because I only called GPIO.cleanup() when I was turning them off. 

I accept that I was completely wrong. I still don’t know what is going on with octoprint though. 

I wrote a script to test the behavior outside of octoprint and discovered my error. When run on an octoprint instance it doesn’t work, but with a standard raspbian install it does. 

I’m truely sorry.

vitormhenrique, 2019-01-31:
No worries @bzeeman, 
You have tried a clean installation haven't you? Are you using the standard octopi image?
Any other software installed that could take control of the GPIO?
It's still very strange.... I would  format the SD card, install the octopi image and test again... I did not wanted to come up too rush on my last message, sorry if it did.

Theoreet, 2019-10-29:
Having exactly the same issue, running the standaard octopi image and only the enclosure plugin.

Probably something did go wrong with the raspberry (dist)upgrade/updates i did ?