vitormhenrique/OctoPrint-Enclosure #191: Auto Shutdown before gcode commands execute

zeroprime790, 2018-09-25:
Hello,

Recently I installed a relay to control my printers power with octoprint. I enabled the auto shutdown but the enclosure plugin shuts down the printer before the "After print job completes" gcodes have a chance to be executed. As result the hotend remains in the last position, touching the print and melting it.
 
**To Reproduce**
Add output
Select Regular IO
Enable Auto Shutdown
Start a print

Is it possible to run the gcode commands before the auto shutdown is triggered?

[image]
[image]

vitormhenrique, 2018-09-25:
Do you have the plugin updated?
You should be able to create delays on the shutdown.

[image]

Another way to fix the issue is to actually shutdown the printer by using the enclosure specific g-code commands at the end of your custom g-code.

zeroprime790, 2018-09-25:
I see, can you please specify the format? If I want to delay the shutdown for 5 minutes, I will just enter the number 5 there? Or 5 means five hours?

zeroprime790, 2018-09-25:
Ok, I get it now. Thank you for your help 👍

closed 2018-09-25 by zeroprime790