vitormhenrique/OctoPrint-Enclosure #194: Set by default an enclosure température

Lepterois, 2018-10-06:
Hi!
It would be great or I don't know to do it, to be able to fix by default a value for the temperature control of the enclosure.
Each time, I need to open the Enclosure plugin tab, and set a temperature value. After a reboot or A print stop, I need to do it again.
I don't find in the configuration tab a field where to set by default the temperature.
Or Is it possible to do it in the yaml?
Thanks in advance

vitormhenrique, 2018-11-08:
If you enable auto-startup on the settings for the temperature control, on the very bottom of that output it's going to appear a "default value" field.
[image]

closed 2018-11-08 by vitormhenrique

Lepterois, 2018-11-08:
Hi!
Yes I've done that but each time I switch off the printer, this value become empty.
Thanks