vitormhenrique/OctoPrint-Enclosure #196: PWM Error Duty Cicle value needs to be between 0 and 100!

glauberbiro, 2018-11-02:
Hi there.

I'm using enclosure plugin for a few weeks and when I try to set a simple output IO command to the GPIO ports everything works fine. 

Now I want to change the IO output to a PWM signal, sometimes I can set the new PWM duty cicle value, without any problems, but sometimes doesn't works, so I need to restart octoprint and works again. When I can't set the the message appers "Duty Cicle value needs to be between 0 and 100!" for whatever value.  Could someone help me?

[image]

vitormhenrique, 2018-11-08:
I can't reproduce this on my rpi, I would need to remotely connect to your setup to troubleshoot, I don't have time to look into this right now, but I'll investigate when possible.
I'm going out on vacations for 30 days and so honestly I'll only have time next year...

otherguy, 2018-12-26:
@vitormhenrique very likely -- from the screenshot -- there is no value in the text box. If you enter 100 and press the checkmark button, the text box is cleared. Pressing the checkmark button again produces the error above.

glauberbiro, 2019-01-27:
Hi @vitormhenrique, sorry for the late answer. Yes I still have the same.

[image]

vitormhenrique, 2019-01-27:
I can not reproduce the issue, do you wanna do me a favor? I would need to log into your octoprint so I can trouble shoot, you can send me an email (my email is on my github page) with the ip address of your octoprint and username and password. 
Also, can you confirm that you are pressing the button only once after entering data, after that, the number will be grayed out, so if you press it again you will see the message above. So again, type a number, press the button only once, the number will turn grayed out, after that you should NOT press the button anymore unless you enter a new value....

Thomas128, 2019-03-27:
I have the same exact problem.

blacknose, 2019-07-11:
Same is happening here. You can reproduce the bug with the following steps:

1- set the pwm duty and confir twice or more

looks like the value, despite been showed at the control is not really send as a parameter. I can fix it if you want some help with the issues.

vitormhenrique, 2019-07-11:
the first time it sends the pwm, and sets the background of the input to the value that was sent, you are only supposed to send it once, the value remains " grayed out"  to show the current value of the pwm.

blacknose, 2019-07-11:
Yes! Sorry! I haven't realized it was greyed out.

vitormhenrique, 2019-07-11:
@blacknose sem problemas!

closed 2019-07-11 by vitormhenrique