vitormhenrique/OctoPrint-Enclosure #197: 10k thermistor

guille50, 2018-11-08:
Hi, this project looks very interesting but looking at the documentation I didn't find any clue about using      regular 10k thermistors like this https://sc01.alicdn.com/kf/HTB1wtfTi_TI8KJjSsphq6AFppXaQ/10K-3977-NTC-Thermistor-For-Hair-Straightener.jpg_350x350.jpg.

I think this will be a very interesting feature because you can put the thermistor directly on your printer board in order to get more precise temperatures.

Is that feature in your road map in the short term if is not I will be pleased to write that code but will you accept my merge request?

 Regards

vitormhenrique, 2018-11-08:
Raspberry PI don't have ADC's, they can not read analog voltages directly, it would require adding an external ADC. The plugin already support plenty temperature sensors, some cost last than 2 to 3 bucks, that defeat's completely adding an ADC to the PI to read thermistors.

closed 2018-11-08 by vitormhenrique

guille50, 2018-11-08:
OMG! I forgot that part, recently I was working with an Arduino doing
analog temperature reads, that's the reason why I was confused

Thanks!