vitormhenrique/OctoPrint-Enclosure #198: Neopixel direct not working from browser, but same command taken from log file does.

Speedster2001, 2018-11-27:
**Describe the bug**
A clear and concise description of what the bug is.

Tested both an IO and Neopixel in a Putty/SSH with python script before using Enclosure and they work.

I have created Neopixel direct in Octoprint/Enclosure plugin and it does NOT work 
[octoprint (3).log](https://github.com/vitormhenrique/OctoPrint-Enclosure/files/2622166/octoprint.3.log)
from the browser (A regular IO does work), in the log file I see
this:
018-11-27 22:36:21,833 - octoprint.plugins.enclosure - INFO - Sending neopixel cmd: sudo python /home/pi/oprint/lib/python2.7/site-packages/octoprint_enclosure/neopixel_direct.py 18 8 128 60 33 227 10

I pasted this command in Putty/SSH on the raspberry and then it works.
sudo python /home/pi/oprint/lib/python2.7/site-packages/octoprint_enclosure/neopixel_direct.py 18 8 128 60 33 227 10

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.
I hoped it would work from the browser and I see it in the log file, but it looks like the command is not executed (rights?)

**Additional context**
Add screenshots of your settings screen, so I know how to recreate the data. Also enable the debug information under the advanced section of the plugin, reproduce the issue and attach the [octoprint log](https://discourse.octoprint.org/t/where-can-i-find-octoprints-and-octopis-log-files/299) here.

[octoprint (3).log](https://github.com/vitormhenrique/OctoPrint-Enclosure/files/2622171/octoprint.3.log)

Best Regards, Richard Hoogeveen, Netherlands

vitormhenrique, 2018-11-28:
Try disabling sudo under advanced sextions od the plugin.

Speedster2001, 2018-11-28:
Hi,

Thanks for your reply !

I have tried that. Same thing, not succesfull.

Thanks, Richard

Speedster2001, 2018-11-28:
Hi,

p.s.
and if I do not include sudo, the relay on the IO also stops working. It
seems to be needed.

BR, Richard

hoogeveen.richard@gmail.com>:

Speedster2001, 2018-11-28:
Hi Vitor,

I tried a lot, wrote a new image file of Octoprint, again tested the IO with a relay and Neopixel outside "Enclosure", they both work.
Again the neopixel does not run within the Octoprint and I do see the command come over in the log file:
sudo python /home/pi/oprint/lib/python2.7/site-packages/octoprint_enclosure/neopixel_direct.py 18 8 255 108 22 191 10
or without sudo.

-------

The same command runs ok on the pi itself from a SSH.

Without sudo I get:

pi@octopi:~ $ python /home/pi/oprint/lib/python2.7/site-packages/octoprint_enclosure/neopixel_direct.py 18 8 255 25 259 25 10
Can't open /dev/mem: Permission denied
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python2.7/site-packages/octoprint_enclosure/neopixel_direct.py", line 21, in <module>
    strip.begin()
  File "build/bdist.linux-armv7l/egg/neopixel.py", line 106, in begin
RuntimeError: ws2811_init failed with code -5 (mmap() failed)

Speedster2001, 2018-11-28:
I have made an Arduino micro with the ino code provided, turn sudo off, connected the I2C, I have seen address 4 on I2c bus and made a **Indirect** Neopixel. It works !!

closed 2018-11-28 by Speedster2001