vitormhenrique/OctoPrint-Enclosure #199: Recommendation for temp probe?

ainuke, 2018-11-29:
Sorry, not a feature request but a recommendation request. I couldn't find another place to ask...

I've run up against some issues with keeping my Prusa in my unheated basement during the oncoming New England winter: 10-15 degree ambient temps are creating printing issues I wasn't having two months ago. I have an enclosure that I'll be heating soon and have really liked the light control that the Enclosure Plugin has brought me. 
Time for temperature. 
What is the most reliable/best-supported/ubiquitous temp sensor for use with this plugin? I can see a list of ones that are supported in the settings, but have little experience with such things and don't know the difference. Is there an "easiest"? 

One of my reasons for buying a Prusa was the reputation of less fussing, more printing; I appreciate your efforts to make that as much a reality with enclosure design!
Thank You!
Erik

bumbalu, 2018-12-07:
I have only used DS18B20 and it works as expected and was set up exactly as recommended. As far as I remember it was not recommended for very high temps and I don't have temps above 75C (mostly print abs at 50C) in my enclosure, so it's OK for me.

vitormhenrique, 2018-12-20:
I suggest DS18B20 as well.

closed 2018-12-20 by vitormhenrique

psi-4ward, 2019-01-10:
Did you test how fast the DS18B20 adopts the real temperature? I didn't used this one but some others which needs some (more) time to adjust the measuring.

vitormhenrique, 2019-01-10:
DS18B20 is SUPER fast :)

psi-4ward, 2019-01-10:
If it's so fast as your answer it's perfect. Thank you!

psi-4ward, 2019-01-10:
Ah, forgot to ask which version of the sensor do you use? The plain one for PCB mount or some kind with wire and enclosure? Think this makes a big difference