vitormhenrique/OctoPrint-Enclosure #200: Print resumes before "resume" button is pushed

bumbalu, 2018-12-05:
Hello, Vitor, hi, everybody.

The bug reproduces only if Enclosure plugin (EP) is enabled. If EP is disabled the bug never happens. 

** The bug **
I have inserted @pause command into the G-code to change filament during print. The print pauses at the correct time/point. Changing filament takes up to 2 minutes. Sometimes printer correctly waits those 2 minutes and resumes printing only after "resume" button is pushed in OctoPrint interface. But almost every other time printer resumes to print about 40 seconds after the pause started WITHOUT "resume" button to be pushed (even without pushed accidentally). 

I do not use filament sensor in Enclosure plugin. I use DS18B20 temp sensor, enclosure ventilation fan, heater and led control. 

**To Reproduce**
Take any G-code. 
Insert @pause command at the beginning of any layer. 
Print until printer pauses. 
Wait more than two minutes. With a chance of ~50% printer will resume printing without user to push the resume button. 

**Expected behavior**
Printer resumes only after user pushes the "resume" button in OctoPrint interface. 

** Enclosure plugin settings **
=== Rpi Input ===
Label = Enclosure_temperature
Input type = temperature sensor
Sensor type = DS18B20
Sensor pin = 4
DS18B20 serial = 28-014e70070002
show temperature on navbar = yes

=== Rpi outputs ===
Output type = temperature/humidity control 
label = enclosure_fan
Id =1 
IO number = 13
autoshutdown = yes
shutdown delay = 1200
temp sensor = enclosure_temperature
value deadband = 2 

Output type = temperature/humidity control 
label = enclosure_heater
Id =2 
IO number = 5
autoshutdown = yes
shutdown on failed or cancelled = yes
shutdown delay = 0
temp control type = heater
temp sensor = enclosure_temperature
value deadband = 2 
maximum temperature = 70

Output type = regular IO
Label = LEDs
Id = 3
IO number = 27 
start with server = yes
autostartup = yes
Startup delay = 0
show button on navbar = yes

** Octoprint scripts **
=== after print job is paused script ===
{% if pause_position.x is not none %}
G91 ; set relative
M83 ; extruder relative
G1 Z+5 E-5 F4500 ; retract 5 mm and move 5 mm up
G90 ; set absolute
M82 ; extruder absolute
G1 X10 Y-10 ; go to safe rest position to change filament
M300 S300 P2000 ; make beep for 2 sec
; M117 change filament
{% endif %}

=== before print job is resumed script ===
{% if pause_position.x is not none %}
M83 ; relative extruder
; prime nozzle
G1 E-5 F4500
G1 E5 F4500
G1 E5 F4500
G4 S2
M82 ; absolute E
G90 ; absolute XYZ
G92 E{{ pause_position.e }} ; set E to where it was before pause
; move back to pre-pause position XYZ
G1 X{{pause_position.x }} Y{{pause_position.y }} Z{{pause_position.z }} F4500
; reset to feedrate before pause if available
{% if pause_position.f is not none %}G1 F{{ pause_position.f }}{% endif %}{% endif %}

** octoprint.log , part at the pause moment *** 

2018-12-04 19:08:10,957 - octoprint.plugins.enclosure - INFO - DS18B20 result: 32.0
2018-12-04 19:08:10,960 - octoprint.plugins.enclosure - INFO - Sensor Enclosure_temperature Temperature: 32.0 humidity 0.0
2018-12-04 19:08:14,204 - octoprint.util.comm - INFO - Changing monitoring state from "Printing" to "Pausing"
2018-12-04 19:08:15,035 - octoprint.util.comm - INFO - Changing monitoring state from "Pausing" to "Paused"
2018-12-04 19:08:21,917 - octoprint.plugins.enclosure - INFO - DS18B20 result: 32.062
2018-12-04 19:08:21,921 - octoprint.plugins.enclosure - INFO - Sensor Enclosure_temperature Temperature: 32.1 humidity 0.0
2018-12-04 19:08:30,501 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2018-12-04 19:08:32,876 - octoprint.plugins.enclosure - INFO - DS18B20 result: 34.687
2018-12-04 19:08:32,881 - octoprint.plugins.enclosure - INFO - Sensor Enclosure_temperature Temperature: 34.7 humidity 0.0
2018-12-04 19:08:32,883 - octoprint.plugins.enclosure - INFO - Turning gpio to control temperature on.
2018-12-04 19:08:32,888 - octoprint.plugins.enclosure - INFO - Writing on gpio: 13 value True
2018-12-04 19:08:43,836 - octoprint.plugins.enclosure - INFO - DS18B20 result: 32.625
2018-12-04 19:08:43,840 - octoprint.plugins.enclosure - INFO - Sensor Enclosure_temperature Temperature: 32.6 humidity 0.0
2018-12-04 19:08:54,796 - octoprint.plugins.enclosure - INFO - DS18B20 result: 31.875
2018-12-04 19:08:54,801 - octoprint.plugins.enclosure - INFO - Sensor Enclosure_temperature Temperature: 31.9 humidity 0.0
2018-12-04 19:08:54,804 - octoprint.util.comm - INFO - Changing monitoring state from "Paused" to "Resuming"
2018-12-04 19:08:54,834 - octoprint.plugins.enclosure - INFO - Turning gpio to control temperature off.
2018-12-04 19:08:54,839 - octoprint.plugins.enclosure - INFO - Writing on gpio: 13 value False
2018-12-04 19:08:57,237 - octoprint.util.comm - INFO - Changing monitoring state from "Resuming" to "Printing"
2018-12-04 19:09:00,556 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2018-12-04 19:09:05,915 - octoprint.plugins.enclosure - INFO - DS18B20 result: 31.25

** serial.log , at the resuming **
2018-12-04 19:08:53,173 - Recv: T:194.82 /195 B:52.99 /53 B@:0 @:132
2018-12-04 19:08:54,052 - Recv: wait
2018-12-04 19:08:54,259 - Recv: T:194.82 /195 B:52.93 /53 B@:47 @:118
2018-12-04 19:08:54,804 - Changing monitoring state from "Paused" to "Resuming"
2018-12-04 19:08:54,846 - Send: N27360 M83*24
2018-12-04 19:08:54,868 - Recv: ok 27360
2018-12-04 19:08:54,875 - Send: N27361 G1 E-5 F4500*51
2018-12-04 19:08:54,904 - Recv: ok 27361
2018-12-04 19:08:54,911 - Send: N27362 G1 E5 F4500*29
2018-12-04 19:08:54,935 - Recv: ok 27362
2018-12-04 19:08:54,941 - Send: N27363 G1 E5 F4500*28
2018-12-04 19:08:54,963 - Recv: ok 27363
2018-12-04 19:08:54,970 - Send: N27364 G4 S2*104
2018-12-04 19:08:54,978 - Recv: ok 27364
2018-12-04 19:08:55,363 - Recv: T:194.82 /195 B:52.87 /53 B@:44 @:126
2018-12-04 19:08:56,449 - Recv: T:194.71 /195 B:53.11 /53 B@:0 @:133
2018-12-04 19:08:56,976 - Send: N27365 M82*28
2018-12-04 19:08:56,980 - Recv: busy:processing
2018-12-04 19:08:57,153 - Recv: ok 27365
2018-12-04 19:08:57,160 - Send: N27366 G90*22
2018-12-04 19:08:57,168 - Recv: ok 27366
2018-12-04 19:08:57,174 - Send: N27367 G92 E0.0*94
2018-12-04 19:08:57,184 - Recv: ok 27367
2018-12-04 19:08:57,190 - Send: N27368 G1 X133.99 Y126.94 Z3.8 F4500*16
2018-12-04 19:08:57,207 - Recv: ok 27368
2018-12-04 19:08:57,213 - Send: N27369 G1 F360.0*108
2018-12-04 19:08:57,232 - Recv: ok 27369
2018-12-04 19:08:57,236 - Changing monitoring state from "Resuming" to "Printing"
2018-12-04 19:08:57,250 - Send: N27370 M106 S255*100
2018-12-04 19:08:57,271 - Recv: ok 27370
2018-12-04 19:08:57,277 - Recv: Fanspeed:255

My OctoPrint ver. 1.3.9, Enclosure plugin v.4.12, printer firmware Repetier 1.0.3, no Klipper
Otherwise Enclosure plugin works fine: it monitors enclosure temp, switches heater and ventilation according to temperature, switches LEDs. It is the most useful (and loved by me) Octoprint plugin. 
Thank you.

bumbalu, 2018-12-08:
In the meantime I have also tried to use M600 command in G-code. In one try out of three the pause was also prematurely interrupted and print resumed before I pushed the encoder. It happened when enclosure plugin was enabled. When enclosure plugin was disabled pausing with M600 and resuming worked as expected (printing resumed only after I pushed the encoder). This is additional argument that enclosure plugin makes the printer to resume during the pause.

vitormhenrique, 2019-01-27:
Do you have any other plugin installed?
I have no idea why this would happen, I only have marling printers, is the pause command supposed to be handled by octoprint or is it a repetier thing?
I'm going to do some research.

bumbalu, 2019-01-28:
Hello, Vitor. 
I do have several other plugins installed. If needed I can give the list of them later. But Enclosure plugin seem to be the only involved: when I disable it and leave other plugins active the bug is not reproduced. When I enable Enclosure plugin the bug re-appears. 
The @Pause command is handled by Octorpint. When I mention "Pause" and "resume" buttons I mean the buttons of Octoprint interface, not Repetier interface. Repetier is only firmware, it does what Octoprint tells it to do. 
It seems like Enclosure plugin "doesn't know" that print is paused and while during the pause it still monitors enclosure temp and sends signals to heater and ventilation (cooler) those signals trigger Octoprint to resume the print. Sorry but I did not notice if premature resume was contemporal with heater or cooler turning on or off. I think this should be investigated. Or as test you may try to disable temp monitoring or sending those signal during execution of @Pause command. 
Thank you for investigating in this bug. 
I wish you wonderful trip to New Zealand (what I also dream to do some day)!

maukcc, 2019-04-10:
We can concur that this happens. 
We did some tests and the print resumes when our dehumidifier is switched off because the humidity gets below the trip value. (the off to on state does not resume the print)
The same goes for the temperature control.
All other in and outputs of the enclosure plugin have no effect to the pause.
All other plugins also have no effect to the pause

bumbalu, 2019-08-26:
Vitor, have you got any news on this issue? 
Thank you.

Nickelboerni, 2021-04-03:
Have you found any workaround?
except disabling the Temperature Control and Hum Control?