vitormhenrique/OctoPrint-Enclosure #202: Setting BME280 sensor doesn't work

hozakjiri, 2018-12-07:
Hello guys, 

Please, can you help me? I have octoprint 1.3.10rc4 on Octopi 0.15.1. My problem is that the sensor BME280 doesn't work via putty it is working. The library is correct but in octoprint is temperature and humidity are zero.  In enclosure plugin i have address 0x76, if i write to command i2cdetect -y 1 after i can see my sensor on address 76. I don't know why doesn't work.  If I write command sudo python BME280,py 76 to putty so he show temperature and humidity...23,2 celsius and humidity 58

Thanks

sebastian-heiden, 2018-12-09:
I have the same problem with th DHT22

sebastian-heiden, 2018-12-09:
Hey I just found the solution to my proble.m
I had to disable the "use sudo" option in the "Advanced Options" tab. This might work also for your problem.

vitormhenrique, 2018-12-17:
of course it does, it's on the troubleshooting guide... and plenty other open issues already opened here.

closed 2018-12-17 by vitormhenrique