vitormhenrique/OctoPrint-Enclosure #205: WS2801 LEDs not turning off /

trendelkamp, 2019-01-05:
**Describe the bug**
When setting the RGB value to rgb(0,0,0) through the Enclosure Plugin the LEDs do not turn off. Instead color is changing to red which is not so bright. When using a shell script the LEDs turn off.
I do use WS2801 LEDs

**Expected behavior**
That the LEDs turn off
[ledoff.txt](https://github.com/vitormhenrique/OctoPrint-Enclosure/files/2729568/ledoff.txt)

**Additional context**
I added a shell script which turns the LEDs off as expected.

vitormhenrique, 2019-02-11:
Gave you tried to use gcode commands instead of shell scripts?

For example:

So create a button and set it to g-code.
Than for the gcode you will use something like this:
`ENC O[id] R[red value] G[green value] B[blue value]`

Where you can get the id created by the output on the settings.

So using:
`ENC O1 R0 G0 B0`

Should turn the neopixel off if you got it working from the UI.

vitormhenrique, 2019-02-26:
Hello, could you please see if the gcode command above works?
Also, what script are you using to turn the led's off?

vitormhenrique, 2019-03-19:
I'm closing the issue because of lack of response from the author.

closed 2019-03-19 by vitormhenrique