vitormhenrique/OctoPrint-Enclosure #207: State ON when Button HELD. OFF when RELEASED

LeonVolta, 2019-01-09:
Not 100% sure how to clearly write this. However I will explain my situation and hopefully that helps.

I have a lighting set up in a control box for my 3D Printer and Enclosure which doesn't always need to be on, however I would like to rig up a Microswitch or Momentary Button so automatically turn the light on when I open the door to the control box. When the door is closed again, I would like the Light in the Control box to turn off.

I have tried rigging this up via GPIO to Normally Closed and Common on a Microswitch however if the trigger action is set to 'TOGGLE' it requires pressing and releasing the Microswitch twice. If I set two events on the same GPIO the lights freak out and turn themselves on automatically after a few seconds, then off, etc. Unfortunately I cannot use another GPIO Pin as I have no spares.

This random video I have found is similar to my situation. When the button is held, the LED lights up, when it is released, the LED goes off. https://www.youtube.com/watch?v=DObfgow8ARo

Thank you.

LeonVolta, 2019-01-29:
Awesome, thank you Vitor. Unfortnetely I am unable to test it however next time I have downtime to test it I will require the Microswitch's connection to the Pi. Thank you very much for the change and for the awesome plugin.

y8s, 2019-02-26:
I'm trying to set up the same thing--a door switch that turns on the lights.

My output turns on the lights when printing starts and off 1 hour after printing ends. I want it to also turn on when the door is open and, if no print is going, turn off when the door is closed.

I can get the pi to read the digital state of the switch properly with "gpio read" and I can get the switch to act as a toggle and I can get the lights to come ON when I open the door. But I can't add the dupilcate input to trigger "low" on "falling edge" for the close.

vitormhenrique, 2019-02-26:
I did not commit the code yet, I'm going to commit the changes and you will need to re-install the plugin.

vitormhenrique, 2019-02-26:
@y8s I fixed and committed the changes, but keep in mind that you can come up with conflicting outputs with two different outputs. 
For example you set the light to turn off after 10 minutes of the start, and you open the door at 9 minutes mark, it will still shut the light off when you reach 10 minutes even with the door open.

closed 2019-02-26 by vitormhenrique