vitormhenrique/OctoPrint-Enclosure #210: HELP! Unable to execute shell script via plug in.

FrisbeePuppy, 2019-01-28:
Hello. I am trying to call a shell script via the Enclosure Plugin. I selected the "shell script" bubble and typed in the name of my script along with assigning it to a button on the UI to test the function. unfortunately nothing happens when I press the button. I am able to successfully call the script in the terminal window via ssh but not with the plugin. Do I need to include the path to the script for this to function correctly?

vitormhenrique, 2019-01-28:
You need the full path, and can not use shortcuts like “~” for the home directory.

closed 2019-01-28 by vitormhenrique

FrisbeePuppy, 2019-01-29:
Thank you for taking the time to assist me.

You were right. It seems I incorrectly entered the wrong path. After correcting this, I tried again and I unfortunately encountered another problem. Upon calling the script from the plugin it instantly halts the terminal, octoprint becomes unresponsive to input and the script fails to execute.

The script is a simple web request to the API to send a gcode as a test.

Again I appreciate the help.

vitormhenrique, 2019-01-30:
Scrips need to execute and exit, if they don't they will hang octopi. 
They are only meant to do something once when the button is pressed. Can you post your script here?