vitormhenrique/OctoPrint-Enclosure #212: support adafruit am2315

jhoblitt, 2019-02-03:
It would be great to be able to use AM2315 over i2c.  Adafruit does have a library https://github.com/adafruit/Adafruit_AM2315  Would a PR adding support for this sensor be accepted?

jhoblitt, 2019-02-03:
https://learn.adafruit.com/am2315-encased-i2c-temperature-humidity-sensor/python-circuitpython

jhoblitt, 2019-02-03:
> At this time this sensor does not appear to work well on the Raspberry Pi. We’ve seen mixed success, but YMMV. 
Great -- I've had one sitting around for awhile...

vitormhenrique, 2019-02-04:
What I want to do, instead of adding individual sensors (because every time, someone wants a different sensor) it's to create a generic "python" input, where it would be easy to extend the plugin to any temperature input.
That is on my top to do list, and I want to work on if on the next few weeks.