vitormhenrique/OctoPrint-Enclosure #213: IFTTT notification

petitxef, 2019-02-04:
Hi Vítor,

I have a BME280 sensor and  I’ve configured the IFTTT well (I receive the notification if I start a print)

But i’m Trying to configure the temp alarm (if the temp exceeds more then 10, trigger the IFTTT) but nothing happens (the enclosure is at 24 degree)

I attach the configuration on plugin, maybe I did wrong?

Thanks in advance 

[image]
[image]
[image]

vitormhenrique, 2019-02-11:
@petitxef nothing wrong on your configuration, the current implementation " requires"  you to have an output on the "controlled io". That is why you are not getting notifications, for now just create a dummy output on any unused gpio. 
I'll improve this on the next version so the controlled io is not mandatory.

petitxef, 2019-02-11:
Thank you Vitor for the information!

vitormhenrique, 2019-02-12:
I'll keep it open so I remember to implement on the future... :)