vitormhenrique/OctoPrint-Enclosure #214: Help with Shell Script

BarendDS, 2019-02-10:
Hi there,

I managed to get the neopixels working but now I would like to add a button using Shell Script  to turn them off.

I tried all sorts of combinations but I cant get the Shell Script to work.

- python  /home/pi/oprint/lib/python2.7/site-packages/octoprint_enclosure/neopixel_direct.py 18 44 255 0 0 0 10
- /home/pi/oprint/lib/python2.7/site-packages/octoprint_enclosure/neopixel_direct.py 18 44 255 0 0 0 10

when I enter  sudo python /home/pi/oprint/lib/python2.7/site-packages/octoprint_enclosure/neopixel_direct.py 18 44 255 0 0 0 10 in SSH it works so I'm stuck here.

Can you help me with the command that will work?

Tx in advance.

Kind regards,

Barend

vitormhenrique, 2019-02-10:
The best way to customize buttons and enclosure actions is to use octoprint enclosure specific gcodes.
So create a button and set it to g-code.
Than for the gcode you will use something like this:
`ENC O[id] R[red value] G[green value] B[blue value]`

Where you can get the id created by the output on the settings. 

So using:
`ENC O1 R0 G0 B0` 

Should turn the neopixel off if you got it working from the UI.

BarendDS, 2019-02-10:
Tx Vitor, I will definetly use this!!
Is it also possible to run a python script from the Shell Script functionality?
I might want to mod the neopixel_direct.py to do for example a knight rider routine.
Tx in advance.

vitormhenrique, 2019-02-10:
It’ definetly possible, but I hightly discourage you from doing so.
Especially because you wanna do animations, shell scripts were not designed
for that, they need to excecute and exit because fast, if they don’t they
will lock out octoprint.

BarendDS, 2019-02-10:
yeah I read that somewhere, too bad though, would be nice to have some colorful animations.

Tx for the info!!

vitormhenrique, 2019-02-11:
The best way if you want do ad animations would be to have an arduino controlling the neopixels, have all the logic on it, making animations and whatever else, and connect it using i2c to the raspberry pi with octoprint, octoprint would send data to the arduino when it needs to show a color, or animation and the arduino would handle.
You would need to customize the arduino example that I have:
https://github.com/vitormhenrique/OctoPrint-Enclosure/blob/master/extras/octoprint_neopixel/octoprint_neopixel.ino
And also customize the interface on octoprint to send animatios....

closed 2019-02-11 by vitormhenrique