vitormhenrique/OctoPrint-Enclosure #215: Unable to set target temperatur

b3ster, 2019-02-12:
First of all, great product I really enjoy your work.
Im using a self build case with a cooler and heater build in, controlled via gpio‘s from pi and temp sensors. My problem is that I’m unable to define the target temrature anymore, after upgrading I can’t find this option anymore. 
Hope you have some advice for me ?

The values marked on the screenshot can be changes at runtiome but no option to changes this persistent

[image]

sajovicd, 2019-02-12:
You can set temperature on setup page.

vitormhenrique, 2019-02-12:
as @sajovicd said, if you have autostartup enabled on the settings you will be able to set a "default" temperature for the heater/ cooler.

closed 2019-02-12 by vitormhenrique