vitormhenrique/OctoPrint-Enclosure #217: I can't seem to get Shell Script to work.

trbolexis, 2019-02-16:
I'm trying to run a shell script from a button to shut off USB's and another to turn them back on.

Currently, after powering my printer down, the screen/electronics stay 'on' because they are being powered by the Pi. I've been able to turn the USB ports on and off with the following commands in ssh.

- Turn USB Off: echo '1-1' |sudo tee /sys/bus/usb/drivers/usb/unbind
- Turn USB On: echo '1-1' |sudo tee /sys/bus/usb/drivers/usb/bind

I've turned these into two separate commands: powerUSBon.sh and powerUSBoff.sh placed in /home/pi/scripts/

In SSH, if I send /home/pi/scripts/powerUSBon.sh or /home/pi/scripts/powerUSBoff.sh the action work correctly. The USB ports turn on/off corresponding to the command.

powerUSBon.sh:
[image]

powerUSBoff.sh:
[image]

After failing miserably to get this to work, I setup python scripts that call the shell script: 
on.py and off.py ::: These have both been placed in /home/scripts

::::::::::::::::::::::::::::

on.py
```
import os
os.system("/home/pi/scripts/powerUSBon.sh")
```
[image]

off.py
```
import os
os.system("/home/pi/scripts/powerUSBon.sh")
```
[image]

Calling `python /home/pi/scripts/on.py` or `off.py` works to activate the script and turn the USB on or off (respectively).

I'm at a loss, and I'm certain I'm overlooking something so I can't seem to get it to work. I have no prior experience with Linux, or Python, or Shell Scripting --- this is all trial and error after a LOT of reading online today.

Please advise.

sajovicd, 2019-02-16:
Did you check sudo in advance setting?

trbolexis, 2019-02-16:
Yes.

[image]

trbolexis, 2019-02-16:
I was able to get this to work by using the `python /home/pi/scripts/on.py` or `python /home/pi/scripts/off.py' within Enclosure plugin.

The last thing that I did was `chmod 777` permissions for `/sys/bus/usb/drivers/usb/bind' and `/sys/bus/usb/drivers/usb/unbind'

[image]

Without setting permissions I kept getting 'permission denied' errors from the enclosure plugin.

::::::::::::::

I can also get it to work now after changing permissions for `powerUSBon` and `powerUSBoff`, respectively by entering '/home/pi/scripts/powerUSBon.sh` or '/home/pi/scripts/powerUSBoff.sh` within Enclosure Plugin settings. 

Lastly, I modified my two scripts to mask the "1-1" echo output:
[image]

[image]

Now Enclosure Plugin displays a nice confirmation message about the status:
[image]

[image]

[image]

I think the key was permissions settings... perhaps there is a more appropriate/elegant way to do this. 

Please let me know.

trbolexis, 2019-02-16:
Next up: I need to figure out a way to make 1 button turn on/off USB and turn on/off GPIO in at the same time to make printer shutdown a 1 touch process.

Perhaps this can be done with additional python scripting or Enclosure GCODE? ...

vitormhenrique, 2019-02-18:
Hello @trbolexis glad that you got it working, yeah, it was definitely permission! there is no other way around! I need to better document this on the wiki or read-me.
Yes, the best way to do multiple stuff on one button event is definitely using enclosure GCODES. 
You should still see the return message even by running the shell script via gcodes!

closed 2019-02-18 by vitormhenrique

sfgabe, 2020-01-08:
> Next up: I need to figure out a way to make 1 button turn on/off USB and turn on/off GPIO in at the same time to make printer shutdown a 1 touch process.
> 
> Perhaps this can be done with additional python scripting or Enclosure GCODE? ...

In case anyone is looking for this, I have a sample "Toggle" power switch on GitHub here: https://github.com/sfgabe/custom_scripts/blob/master/octoprint_relay_toggle.sh

I'm using a ramps board with 2 power sources, so I wanted a one touch on/off. For some reason I had to set it as a service to work correctly, but follow the info in the file and it should work.