vitormhenrique/OctoPrint-Enclosure #218: PWM frequency of 25000 Hz shows 5800 KHz on oscilloscope

hixfield, 2019-02-18:
**Describe the bug**
I enabled PWM output on different pins (3, 18,12) for an inputted frequency of 25000 Hz (25 KHz).
When I measure the output I can see the PWM signal but only at 5800 Hz (5.8KHz). I have tested on different output pin, thinking that maybe one of them would have hardware PWM and the others would be software PWM that could not get to the requested 25KHz frequency, but all are the same.

**To Reproduce**
Steps to reproduce the behavior:
1. Enabled output PWM
2. Set to 25000 Hz
3. Measure output on oscilloscope

**Expected behavior**
I would expect an output frequency of 25KHz

**Additional context**
[octoprint.log](https://github.com/vitormhenrique/OctoPrint-Enclosure/files/2876623/octoprint.log)
[image]
[image]

vitormhenrique, 2019-02-18:
I think raspberry pi has a limit of 19.2MHz.
Anyway, this might also be a limitation on rpi.gpio on python.

hixfield, 2019-02-18:
19.2MHz? Thats approx 1000 higherthen the 25KHz I need 😎

vitormhenrique, 2019-02-18:
Lol! Maybe lower limit? Idk...

Have you tried different values?

hixfield, 2019-02-18:
I tried 25Khz and then 1Khz. I am driving a noctua fan, 4 pin, which following spec should get a 25KHz pwm signal however I noticed  it also works on 1Khz... doing research I see a lot of differt libs for pwm on the pi. 

https://raspberrypi.stackexchange.com/questions/54150/what-is-the-maximum-frequency-of-a-pulse-width-modulation-signal-for-the-raspber

I wil write a manual py script to test isolated and let you know

vitormhenrique, 2019-02-18:
I'll test it out on my oscilloscope tonight, I saw a warning on your log related with the pwm, maybe delete and create it again?