vitormhenrique/OctoPrint-Enclosure #220: NeoPixel Direct not working.

AmauryBJ, 2019-02-21:
Hello 

Octoprint with PI2
Pin 18
NeoPixel not working, no info in log.
When trying to execute the script manually in ssh i get : neopixel module not found.

Thank you

vitormhenrique, 2019-02-21:
hare you read the read-me followed he instructions and installed all necessary libraries?

AmauryBJ, 2019-02-21:
i do but on adafruit website they are using python3 and you script is using python2. They say its not compatible with python2

AmauryBJ, 2019-02-21:
and thank you for your quick answer ;)

vitormhenrique, 2019-02-21:
apparently they updated their installation guide, you need to install this:

https://github.com/jgarff/rpi_ws281x

AmauryBJ, 2019-02-21:
Thanks,
i clone this and run the commands in the python folder : 

sudo apt-get install python-dev swig
python ./setup.py build 

During build i get : /usr/bin/ld: cannot find -lws2811

AmauryBJ, 2019-02-21:
i also tried to execute : pip install rpi_ws281x
Execute with success but same error when launching the test neopixel script : no module name neopixel

vitormhenrique, 2019-02-21:
you have to get neopixel working by itself first, than try the script, the enclosure plugin can not work if the libraries are not working.... 

Test the library out of the enclosure, after you get it working I can help you with setting up on the plugin, before sorry but there is not much I can help. I can only support issues with the plugin...

Also, check this: https://github.com/vitormhenrique/OctoPrint-Enclosure/issues/172#issuecomment-410864599

AmauryBJ, 2019-02-22:
I am testing the library without the enclosure plugin, simply in ssh.
But the fact is that your way of using it is using python2, the official library is using only python3 ( Adafruit documentation ).
So someone who want to install a brand new rasperry pi, with octoprint, and then use neopixel with your enclosure plugin as no documentation at all to do it.
I understand that you only support issue with your plugin, but for neopixel, python2 library installation is 95% of the process of making it working. So maybe it can be a good think to include a documentation to explain the step which are not the official ones.

AmauryBJ, 2019-02-22:
By the way, i test neopixel with the official documentation and python3, it's working like a charm

AmauryBJ, 2019-02-22:
The thing is, i can't find the neopixel library working on python2, all the search a made ended on CircuitPython NeoPixel that is only compatible with python3.

vitormhenrique, 2019-02-26:
Uninstall the plugin and install it again, I updated the script to work with the new rpi_ws281x.

If your setup does not have pip install pip:
`sudo apt-get install python-pip`

Install the required library:
`sudo pip install rpi_ws281x`

rpi_ws281x really needs sudo, and you need to setup up so your rpi does not ask for a password for sudo, so run:

`sudo visudo`

and add `pi ALL=(ALL) NOPASSWD: ALL` to the end of the file.

Also backlist the audio kernel:

`sudo nano /etc/modprobe.d/snd-blacklist.conf`

add:

`blacklist snd_bcm2835`

I just tested this on a clean environment and got a strip working...

closed 2019-02-26 by vitormhenrique