vitormhenrique/OctoPrint-Enclosure #222: Simple RGB strip led (color controle)

devildant, 2019-02-25:
Hello,
I have standard RGB led strip, and I wish I could change the color of these since your plugin enclosure.
in my research I found this tutorial, but it is necessary to connect the band led to 3 gpio for the RGB (see link and diagram below)
would it be possible to integrate this option in enclosure?
an option led strip with a color picker :)
[image]
https://medium.com/@danidudas/how-to-connect-rgb-strip-led-lights-to-raspberry-pi-zero-w-and-control-from-node-js-70ddfec19f0b

vitormhenrique, 2019-02-25:
Very few people use those strips, as "smart" LED's (like neopixel) are really cheap now, especially because RGB strips require external transistors to drive them. so I do not plan to integrate this to the enclosure plugin. 
You can add three PWM outputs and control each color individually already, and if you want, create gcodes buttons with pre defined colors, but I'm not going to implement a color picker for it.
That being said the project is open source, anyone can clone the repository and make the changes if more people are interested enough.

closed 2019-02-25 by vitormhenrique

devildant, 2019-02-25:
Ok, 
by uninstalling and reinstalling your plugin, I found an option led strip, and it seems to work with the module below, I should be able to use my led strip. I have good? :)

https://fr.aliexpress.com/item/Full-color-RGB-LED-Strip-Driver-Module-For-Arduino-Can-Be-Cascaded-Blue-Black-Hot-New/1831927656.html

vitormhenrique, 2019-02-25:
As I said, using is one thing, you can use it, but it will not offer a "color picker"  as it does for neopixel control.

devildant, 2019-02-25:
mmm ok, 
I looked at the option available with the strip led there is however a color picker.
[image]
I made this request because I already have a led strip, and the price of neopixel is relatively high ($ 70 for 4 meters, yes I have a large enclosure)
thank you in any case for answering me

vitormhenrique, 2019-02-25:
neopixels are really bright, I bet that 1 strip is more than enough. It can probably illuminate a whole room.
using 3 pwm outputs on each "color"  of your strip would work on your setup with your strip.

devildant, 2019-02-25:
ok thx :)