vitormhenrique/OctoPrint-Enclosure #225: Log errors

valepe, 2019-03-13:
Hi,
I use your plugin and very happy with it.
This morning I looked in the octoprint log because it went unresponsive to web calls and I saw a lot of errors from Enclosure plugin.
Every 10 seconds it fires these two lines:

2019-03-13 00:02:57,755 - octoprint.plugins.enclosure - WARNING - An exception of type IndexError occurred on log_error. Arguments:
('list index out of range',)
2019-03-13 00:02:57,756 - octoprint.plugins.enclosure - WARNING - An exception of type TypeError occurred on log_error. Arguments:
("'NoneType' object is not iterable",)

I don't know where to look to resolve the error.
Enclosure plugin:
 - drives a tworelay module (they works)
 - reads from MQ2 smoke sensor (never tryed)
 - reads from DS18B20 sensor (temperature values are fine)
 - reads two push buttons (they works)
 - reads a switch as filament runout sensor (until now never tryed)
and sends notifications via IFTTT (they work).

Can you help me where I can look in?

Thanks

vitormhenrique, 2019-03-13:
Can you attach the entire log? And printscreen of you configuration?

valepe, 2019-03-13:
here the full log file. When I'll come back home I'll send the screeshots. thanks
[octoprint.log](https://github.com/vitormhenrique/OctoPrint-Enclosure/files/2961436/octoprint.log)

vitormhenrique, 2019-03-13:
From what i’m looking what I would try first is to delete all your settings
inside the plugin, save, and create them again.

Check the log after saving a blank settings to see if the error disappear.

valepe, 2019-03-13:
Ok, I'll try that. In the meantime I made the screenshots.
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

vitormhenrique, 2019-03-19:
have you tried to delete all settings? 
did it fix it?

valepe, 2019-03-20:
Sorry for the delay. I'm quite busy in this period.
I did shutdown the system for some days and when I startup again to fix this problem, it was gone.
No more log errors. I don't know what happened.

vitormhenrique, 2019-03-20:
Good!

closed 2019-03-20 by vitormhenrique

fjohannes, 2021-09-06:
I'm having the same issue here. Potentially leading to an extremely lagging Octoprint UI / Octoprint as whole.
Will try with deleting settings and see what happens..
[octoprint-logs.zip](https://github.com/vitormhenrique/OctoPrint-Enclosure/files/7116577/octoprint-logs.zip)

fjohannes, 2021-09-10:
After deleting all settings, the error was gone.
After re-implementing one Output (printer relay) it was still fine. After adding my DS18B20 temperature sensor, the issue reappeared. Maybe the sensor is broken, but it was reading plausible temp values nonetheless. Might change the sensor and test again..

fjohannes, 2021-09-10:
The issue was reappearing. I sadly had to uninstall Enclosure Plugin for now.
One hint: Issue was appearing close after installing OctoKlipper Plugin. Maybe something weird going on between those two plugings? https://github.com/thelastWallE/OctoprintKlipperPlugin

fjohannes, 2021-09-13:
When was trying to access my Octoprint this morning, I had a hard time because it was unresponsive for quite some minutes (even via ssh). After a while, I could access and checked the log, which gave some erors relating some phython2.7 files/folders missing in a directory named octoprint_enclosure.
e.g.:
IOError: [Errno 2] No such file or directory: '/home/pi/oprint/lib/python2.7/site-packages/octoprint_enclosure/static/js/bootstrap-colorpicker.min.js'
Full log file attached.
[octoprint-logs (6).zip](https://github.com/vitormhenrique/OctoPrint-Enclosure/files/7153674/octoprint-logs.6.zip)
How to repair this?
(when uninstalling Enclosure Plugin I choosed the option "including cleanup", btw.)