vitormhenrique/OctoPrint-Enclosure #227: Logging level.

Tirpitz93, 2019-03-28:
Is there a reason you use 
```
if self._settings.get(["debug"]) is True:
self._logger.info()
```

instead of
```
self._logger.debug()
```
?
I have been trying to fix #226.

vitormhenrique, 2019-03-28:
yeah, I created my "own" logging level that is accessible from the enclosure settings under the advanced sections. 
I did this because most users don't would not know how to change the debug level on octoprint and having that info is crucial to troubleshoot and help users out...
So they could simply go on the settings and enable "debug" mode for this plugin only. Octoprint debug level can only be changed on `logging.yaml` and trust me, explaining how to change one simple line on a yaml file can be a challenge. 

If there was a simple way to switch octoprint debugging level I would certainly use 

`self._logger.debug()`

I know, that it's not pretty, but the rest of the code ins't pretty as well, lol

This was my first python project, and there are a lot of things that I don't like, especially that all settings are stored on dictionaries that rely on hardcoded key values. This makes it really hard to extend and understand the code, and the ko bindings on the UI are a pain in the *** to maintain because the data needs to be carefully "handled" when something changes on the "model" dictionary.

I don't know a better way to refactor this, but I my guess is to create a base class and subclasses and somehow make octoprint settings save it.

I plan to refactor the code to implement #169 but I haven't had time on the last few months other than fixing "major" issues when I find them...

Tirpitz93, 2019-03-28:
Pretty sure this is better:
[image]

Tirpitz93, 2019-03-28:
On this topic, where is `debug_temperature_log` set?

vitormhenrique, 2019-03-28:
I don't know how I missed this! definitely better. `debug_temperature_log` is a check box available also under the advanced section, after you enable "debug" on the plugin.

I guess putting everything under the debug is the way to go.

Tirpitz93, 2019-03-28:
> I don't know how I missed this! definitely better. `debug_temperature_log` is a check box available also under the advanced section, after you enable "debug" on the plugin.
> 
> I guess putting everything under the debug is the way to go.

I will have a pull request on the way shortly

vitormhenrique, 2019-03-28:
Thanks for the fixed and changes!

closed 2019-03-28 by vitormhenrique