vitormhenrique/OctoPrint-Enclosure #235: Filament Change issue

user012flo, 2019-04-18:
I use the enclosure plugin for a filament change with a filament runout sensor.
When the filament goes out, the plugin detects the signal from the sensor.
So the settings have to be correct.

But then the printer does not execute my inserted script, but first moves with the hotend to the position at the far right back of the heating bed. And then it executes my script.

My filament change gcode:
M83
G1 E-5
G91
G1 Z10
G1 E5
M82
G90
M600
G91
G1 Z-10
G90

Is it possible to deactivate the movement of the hotend to the far right back?

vitormhenrique, 2019-07-11:
I need more info, like what printer do you have, firmware, debug information, etc etc.. check the wiki on how to open an issue.

energiebrand, 2020-06-11:
This is an old issue but now I can confirm exactly the same behaviour on my Sidewinder X1 on a RPi 3B. Problem is, that changes in the filament change G-Code are not taken by the PlugIn. I checked the YAML and it contains my new code but the PlugIn use the default G-Code script of the PlugIn. And the nozzle moves up 15mm and then moves to the far end X/Y postion as stated above but I have removed this G-Code from the default.
I will show you my G-Code and add some screenshots and a log tomorrow.

energiebrand, 2020-06-13:
So here comes a YouTube video I made to show the behaviour: https://youtu.be/kAksutksZ9w 
Also I made some screenshots of the configuration of the PlugIn as well as I attached the YAML and the log as well as my G-Code without X/Y axis movement. Hope it helps for debugging.

[octoprint.log](https://github.com/vitormhenrique/OctoPrint-Enclosure/files/4774591/octoprint.log)

[G-Code.docx](https://github.com/vitormhenrique/OctoPrint-Enclosure/files/4774592/G-Code.docx)

[YAML.docx](https://github.com/vitormhenrique/OctoPrint-Enclosure/files/4774589/YAML.docx)

[image]
[image]
[image]
[image]

louisferreira, 2020-11-28:
I have the exact same issue.