vitormhenrique/OctoPrint-Enclosure #236: Multiple PWM doesn't work reliably

HaythamB, 2019-04-19:
**Describe the bug**
Setting up 3 separate PWM outputs for RGB 5050 LED strips doesn't change duty cycles reliably, ending up with this in the logs:

octoprint.plugins.enclosure - WARNING - An exception of type RuntimeError occurred on log_error. Arguments:
('A PWM object already exists for this GPIO channel',)

**To Reproduce**
Steps to reproduce the behavior:
1. Create 3 PWM outputs, map to any default GPIO pin
2. Try and change duty cycle normally, and then change another PWM duty cycle

**Expected behavior**
Duty cycle should trigger an output, nothing happens, and log outputs the above for debug.

HaythamB, 2019-04-19:
Tested this out further. Turns out it requires a restart of the OctoPrint service every time you change the PWM frequency. Working properly after restarting the service.

RoCkHeLuCk, 2021-04-30:
I have the same problem, if I change any PWM plugin configuration to, and in the log the message "A PWM object already exists for this GPIO channel", just restarting to get back to work.

jacopotediosi, 2025-05-01:
Hi everyone, the recently merged PR #504 should have fixed this issue. Could you please reinstall the plugin and, if confirmed, close the issue?