vitormhenrique/OctoPrint-Enclosure #238: [F/R]: Switch GPIO when webstream is called

6ffm70, 2019-05-20:
**Problem:
When watching the webstream while it's dark at the printer, I have to manually switch the light on (via the Enclosure plugin). 
Also, automated sending snapshots from a messenger plugin would require to have the enclosure illumination to be permanently switched on.

**Solution:
The plugin shall make use of Octopi's function that the webcam-stream get started, when it is called once or by multiple users. The same trigger can be used to switch a GPIO.

**Bonus Solution:
A popular light sensor can be connected to the Raspi, and pause the illumination feature (via the plugin) when it's not dark.

vitormhenrique, 2019-07-11:
I need to do some research to see if it is possible.

mark-hahn, 2019-07-27:
This would be nice for me and anyone who peeks a look at a print every 15 mins or so.  It would save a lot of clicking around.

6ffm70, 2019-09-30:
> I need to do some research to see if it is possible.

I'm still keen, Henrique! :)

6ffm70, 2020-02-25:
I was browsing the OctoPrint documentation for clues how to make this possible. There's an event called `ClientOpened` mentioned here [http://docs.octoprint.org/en/master/events/index.html#available-events](url), which occurs when a client connects to the webserver. Equally, there's an event available for `ClientClosed`. That's not limited to the webstream tab, but could already minimise the time the camera light is turned on. Could that be easily implemented?