vitormhenrique/OctoPrint-Enclosure #242: PID feedback loop

truglodite, 2019-06-06:
**Is your feature request related to a problem? Please describe.**
No problems per se, just wanting better performance

**Describe the solution you'd like**
Implement a PID feedback loop for fans that are dependent on temperatures. The existing P only control cannot compensate well for external variables such as heater and fan sizes, insulation of enclosure, ambient temperature, bed temperature, etc. 

**Describe alternatives you've considered**
I am currently using a pro micro to control my enclosure with PID loops. It reads GPIO from octoprint to trigger auto, heating, and cooling modes (also an LCD interface).

**Additional context**
It would be great to have the same level of precision without the extra hardware. Unfortunately I don't know py or js, so thos is a fr not a pr.

vitormhenrique, 2019-07-11:
This feature has been requested over and over, and I explained that it is probably an overkill for this implementation. the enclosure has a long period of residence, the temperature changes really slow, compared to the power getting in, so unless you have a reaaaaalllyyyy oversized heater  / fan, i do not believe you need PID control. 
I might add on the feature, but I'm not entirely sold if it is worth. Especially if using relays to control heaters, if you are not modulating the heater element controlling an on / off switch with PID control makes no sense.

truglodite, 2019-07-11:
I don't use pid for the heater since that doesn't even need to be on on my setup except for the initial heatup. After that, the printer is adequate to maintain heat. However, the mixing and cooling fans are much nicer with pid. No on/off stiff breeze inside... just gradual perfection. Also perhaps my exhaust fan is larger that average... a 92mm 2A delta with stator. It is needed to provide enough pressure to push open my intake and exhaust dampers and intake filter. On a typical print the exhaust settles in around 18-30% pwm... in cooldown mode it takes about 4min at 100% to cool from 50C to 25C. 

It may be a special case, and my fan may be in your opinion 'too big'? Anyhow you'd have to see it printing to appreciate it. Many others might be after similar with their requests, and it can be done in arduino easy enough if this is overkill for octoprint. Imho it is not overkill on a machine designed to exploit pid control, but maybe octoprint can't handle it 
1lwithout introducing problems?

vitormhenrique, 2019-07-11:
I see! On your case PID might help and makes sense,  but most people that
asks for PID, want to control a relay board for the heater....
I might implement this on the future, but next release will only be in a
few months.
I’m working on other projects right now and don’t have a lot off free time
for the enclosure plugin.

truglodite, 2019-07-11:
Good to hear! I am in the position of begging as I don't know enough python or java to help with coding. Of course no rush needed since I have an arduino to do the work on my printer right now. If/when pid fans are added I'll be able to simplify my printer a bit. ;)

Poodlepapa, 2020-06-19:
I too would LOVE this.  I have an oversized heater, NO RELAYS! SSR mounted to a heat sink.  Chamber is lined with 1' foam, 1/2 cement board 1/2 drywall.  I also use it to sanitize my masks with heat and uv.  Ill beta test if you like