vitormhenrique/OctoPrint-Enclosure #244: Unable to making work a neopixel ring

ErSedavi, 2019-06-15:
First of all, I've to say that I'm a newbie at this world of programing, so please be patient with me. Somehow I was able to make run a DHT11 when it didn't after some tries but the neopixel ring is killin me.

I'm runing RPi 3B+
A 12 WS2812 LED ring.
Octoprint 1.3.11 - OctoPi 0.16.0 

When I click on the neopixel button on the pluggin tab it shows me an error that saids "_**Color needs to follow the format rgb(value_red,value_green,value_blue)!**_"

I've tried enabling/disabling sudo but nothing happens.
I've tried to change the neopixel_direct.py file as you post on #172 

I've read about test the ring fist out of Enclosure using SSH but have no idea how to do that.

[image]
[image]
[image]
[image]
[image]

[octoprint.log](https://github.com/vitormhenrique/OctoPrint-Enclosure/files/3293335/octoprint.log)

Thanks for your help.

yamanub, 2019-06-16:
This one took me quite a while to figure out as well. I finally got it working by following this guide https://dordnung.de/raspberrypi-ledstrip/ws2812  As soon as I got the LED's working from this guide, I can now activate via the plugin. I have the exact same setup as you and its working on my neopixel 12 ring.

As for the error "Color needs to follow the format rgb(value_red,value_green,value_blue)!" when clicking the check box, first select a color in the window to the left that reads "rgb(255,255,255)", then the checkbox will work.

ErSedavi, 2019-06-16:
Thanks for respond @yamanub 
I'll try your method but without the level shifter because I don't have anyone, if it doesn't work I'll buy one but this would take at least a week.

ErSedavi, 2019-06-16:
Well, after trying you method and runing a test an error appears.
[image]
Similar errors had appear before while trying to run other tests posted on threads about other issues with LEDs.
Looks like something isn't installed well.

yamanub, 2019-06-16:
Sounds like you are missing some things that need to be loaded. You may also try running through this as well. https://learn.adafruit.com/neopixels-on-raspberry-pi/overview

yamanub, 2019-06-17:
Hopefully this helps you. I just setup a new pi 3b+ and the enclosure plugin with a neopixel 12 ring. These are the steps I followed from a fresh install to get a working LED ring via the plugin with neopixel direct.

install octopi and configure as normal
install the octopi update when prompted.
login to the pi via ssh and run the following commands in this order:

`sudo apt-get update`
`sudo apt-get upgrade`
`sudo apt-get install python3-pip`
`sudo pip3 install --upgrade setuptools`
`sudo raspi-config`
select option 5 - Interfacing Options
enable SPI
enable I2C
`sudo reboot`
after rebooting continue with the following
`pip3 install RPI.GPIO`
`pip3 install adafruit-blinka`
`pip3 install --upgrade adafruit_blinka`
`sudo apt-get install build-essential python-dev python-pip unzip wget scons swig`
`wget https://github.com/jgarff/rpi_ws281x/archive/master.zip && unzip master.zip && cd rpi_ws281x-master && sudo scons && sudo pip install rpi_ws281x`
Change directory to rpi_ws281x-master/python/examples where the strandtest.py file is located
`cd rpi_ws281x-master/python/examples/`
`sudo nano strandtest.py`

When you open strandtest.py you need to change the following:

`from neopixel import *`
to
`from rpi_ws281x import *`

also change your LED count to the number of LED's you have on your strip or ring (mine was 12). Save and close strandtest.py

With your strip or ring connected to GPIO 18 you should be able to run: `sudo python strandtest.py` and your LED's will light up and run a sequence specified in the strandtest.py file

As long as it lights up, the last step is to install the enclosure plugin. Reboot octoprint and configure an ouput for neopixel direct in the settings of the plugin, make sure to use GPIO 18 and set the number of LED's, 
Then back to ssh and run the following:
`sudo visudo`
and add `pi ALL=(ALL) NOPASSWD: ALL` to the end of the file, save and close.

After I performed all the steps above, my neopixel 12 ring works via the enclosure plugin. If you continue to have issues you need to check your wiring/connections. I used a level shifter and separate 5v power supply as I am also using relays to control other functions but there is a way to wire without the need of the level shifter, although I cannot confirm this method works using that option.

ErSedavi, 2019-06-18:
Wow! Looks promising.
Thanks a lot for your help, I'll try this afternoon when I arrive home after work.
(7:40 a.m. here I just start)

ErSedavi, 2019-06-19:
@yamanub You're awersome! Works as expected.
Thank you so much for all your help! You've done and incredible effort.
Tonight I'm gonna sleep like a baby full of happiness.

yamanub, 2019-06-19:
Great! and you are welcome! I assumed a level shifter was not necessary with the small neopixel ring but I wanted to be able to add more led's in the future so I installed one on mine.

closed 2019-06-24 by ErSedavi