vitormhenrique/OctoPrint-Enclosure #247: Gcode GPIO

davidcgu, 2019-07-05:
Good afternoon

It will be great to be able to control a GPIO in any direction in/out being able to make a link with Gcode, so for instance make a link on a defined GPIO out that is drive directly on the Gcode with M106 so it works as part cooling fan on a standard Gcode.

vitormhenrique, 2019-07-05:
The plugin can already handle GPIO control over gcode.

Please read the documentation, it uses special gcode commands.

davidcgu, 2019-07-05:
Hola Victor, I read already the docu, could be controlled via Gcode specifying on the gcode the Id of the pluguin, however I mean to be able to make the link between standard Gcode uses and the id of the pluguin so it works out of the box with a standard Gcode without adding anything special on Gcode, That's the example I explained with M106 to drive directly an output

vitormhenrique, 2019-07-05:
I don’t plan adding more gcode control, the current solution it’s really
simple to use and works with the plugin functionality.
You can add on your pre processor on your slicer custom enclosure plugins
so when you slice it will handle the fans or lights or anything else.

If you want to fork the plugin and check how i made the implementation it
will be fairly simple to extend and add the 106 gcode interpretation.

Best regards,

Vitor Henrique

davidcgu, 2019-07-05:
It was just one idea than can be useful, I put this as example but could handle any other marlin features in case your main board don't have enough I/O implementing this could be a straight forward solution without the need to pots processing gcodes, Anyway thanks a lot for your time, great pluguin, no question about it!!!!!!!  amazing job

vitormhenrique, 2019-07-11:
no worries, the problem is that gcode is being  getting more and more customized by manufactures, so if the function already exists i will not be likely to over customize on my plugin.

closed 2019-07-11 by vitormhenrique