vitormhenrique/OctoPrint-Enclosure #249: No temperature data with DHT22

neovanmatix, 2019-07-13:
**Describe the bug**
I've attached a DHT22-Sensor to my Raspberry 3 with OctoPi and the Enclosure-Plugin, but it show's 0° and 0% humidity on the Webinterface.

The DHT22 is connected (like descriped in the screenshots) to 3,3V with resistor, Ground an GPIO 13 (BCM27).
I've installed the Adafruit-DHT-Library and be able to get the temperature on the CLI:
```
pi@octopi:~/Adafruit_Python_DHT/examples $ sudo ./AdafruitDHT.py 22 27
Temp=25.3*  Humidity=46.3%
```

The configured Input seems correct; also the BCM-Pin 27 (i'm using an relay as output for LED-Stripes; also configured with BCM-Pin and it's working).

[image]

[image]

Any idea's, what the cause could be?

vitormhenrique, 2019-07-13:
Disable sudo under the advanced section.

neovanmatix, 2019-07-13:
Hello, 
thank you very much :-) I found the Debugging-Option in the Enclosure-Addon and also found a regarding Logentry in octoprint.log:

> 2019-07-13 22:36:08,335 - octoprint.plugins.enclosure - INFO - Sensor Intern Temperature: 0.0 humidity 0.0
> 2019-07-13 22:36:18,349 - octoprint.plugins.enclosure - INFO - Temperature dht cmd: sudo python /home/pi/oprint/lib/python2.7/site-packages/octoprint_enclosure/getDHTTemp.py 22 27
> 2019-07-13 22:36:18,452 - octoprint.plugins.enclosure - INFO - Dht result:

After disabling sudo it's working now. Thank you! :-)

closed 2019-07-13 by neovanmatix