vitormhenrique/OctoPrint-Enclosure #250: Using reported temperatures as sensor

Th3M1k3y, 2019-07-15:
**Is your feature request related to a problem? Please describe.**
I am looking at a way to add more fan control to my printer, so it can turn fans on and off, depending on if they are needed or not. The microcontroller on the printer only got one spare output pin, which isn't enough.

**Describe the solution you'd like**
Looking at this plugin, it looks like it is very close to be able to do what I am looking for. Ideally I would be able to add a sensor, which would be using the temperature returned.
```Recv:  T:199.89 /200.00 B:60.54 /60.00 @:68 B@:0```
So here I could select "Hotend" and the sensor would use the 199.89 returned by the printer, and same for "Heated bed", where it would use the 60.54 returned.
Then using that input, I could control a gpio output on the raspberry pi, and then have that control the fan.

**Describe alternatives you've considered**
I considered adding a temperature sensor to get the temperature on the hotend heatsink, but it would just be another point of failure, and just basing hotend heatsink fan on hotend temperature, is plenty good.

mark-hahn, 2019-07-30:
> The microcontroller on the printer only got one spare output pin, which isn't enough.

I'm curious as to why this isn't enough.  Turning the fan on and off only requires a single bit.

Th3M1k3y, 2019-07-31:
It isn't enough, because I also need other things connected.